motioneye-project/motioneyeos #73: Can't add remote MotionEyeOS camera to local MotionEyeOS with local camera

chrismelikian, 2015-12-20:
When I try and add a new camera, the drop down initially has Local Camera for the type and the 'Camera' field has 'No Camera' in it because presumably I already have the local camera set up within this MotionEyeOS. When I try and change the 'type' field to 'remote motioneye camera' the URL field is disabled and I can't type the URL into it. 

[image]

ccrisan, 2015-12-20:
What version of motionEyeOS are you running? Are you sure the URL field is actually disabled? I know it looks grayish but normally you should be able to type in an URL. The username field, on the other hand, is disabled and will always be hardcoded to "admin".

chrismelikian, 2015-12-20:
User error, please ignore...

chrismelikian, 2015-12-20:
Hi Calin,

I'm so so sorry for bothering you. The field looked disabled and I'm
certain it didn't let me edit but I retried just now and it worked fine.
I've closed the comment. Apologies again and many thanks for some great
software!

Best regards, Chris

closed 2015-12-20 by chrismelikian

ccrisan, 2015-12-20:
No worries.