motioneye-project/motioneyeos #74: Upload media files not working

yohannver, 2015-12-20:
Hello,

thank you for this great OS !
I am trying the new version which allow us to upload media files on google drive or dropbox.
I configure it out by adding the location (/camera) and the authorization key.
I tried on both but no pictures are uploaded whereas I enabled motion triggered.
Am i doing something wrong ?

Yohann

ccrisan, 2015-12-20:
Have you clicked on "Test Service"? Did it succeed? If yes, has your motionEyeOS recorded any movies or taken any pictures? If yes, can you please attach your motioneye.log file so I can identify any problems that might have occurred?

yohannver, 2015-12-24:
Yes I clicked on "Test service" and it worked for google drive and dropbox.
My motionEyeOS recorded a lot of pictures yes but they are not uploaded.
My configuration is : Raspberry A+, micro SD evo 32go, camera pie.

I got 2 another issues :
- my camera stops streaming on its own and displays this logo very often :
[image]

- the web ui is accessible in my local network at this adress 192.168.1.49. I configured my router by opening the port 9999 from any ips from the internet to this local ip but it doesn't work. I did the same configuration with motionPie and it worked :(

[motioneye.log.log.txt](https://github.com/ccrisan/motioneyeos/files/71815/motioneye.log.log.txt)

Thanks for your help.

ccrisan, 2015-12-24:
The memory of an A+ is not enough. Please upgrade to a 512mb pi board.

closed 2015-12-25 by ccrisan