motioneye-project/motioneyeos #75: "An error occurred. Refreshing is recommended" after activating mail notification

fnordson, 2015-12-20:
As the title says,
everything worked up to the point where I've entered the notification mail address. 
After the reboot, I get the known error and the camera is not aviable anymore.

I've done this twice now and everytime this is the point where it breaks down.
Motioneye seems still to work though since it's uploading pictures to gdrive and sends mails as set up in config.

Here is the related part of motioneye.log:

2015-12-20 19:49:23: [motioneye]     INFO: rebooting
2015-12-20 19:49:23: [motioneye]     INFO: executing "/sbin/reboot"
2015-12-20 19:50:09: [motioneye]     INFO: hello! this is motionEye server 20151206
2015-12-20 19:50:13: [motioneye]     INFO: cleanup started
2015-12-20 19:50:13: [motioneye]     INFO: wsswitch started
2015-12-20 19:50:13: [motioneye]     INFO: tasks started
2015-12-20 19:50:13: [motioneye]     INFO: mjpg client garbage collector started
2015-12-20 19:50:13: [motioneye]     INFO: smb mounts started
2015-12-20 19:50:14: [motioneye]     INFO: server started
2015-12-20 19:54:35: [motioneye]  WARNING: mjpg client connection for camera 1 on port 8081 is closed
2015-12-20 19:57:01: [motioneye]    ERROR: mjpg client timed out receiving data for camera 1 on port 8081
2015-12-20 20:08:04: [motioneye]    ERROR: No closing quotation
Traceback (most recent call last):
  File "home/ccrisan/motioneyeos/output/raspberrypi/target/usr/lib/python2.7/site-packages/tornado/web.py", line 1369, in _stack_context_handle_exception
  File "home/ccrisan/motioneyeos/output/raspberrypi/target/usr/lib/python2.7/site-packages/tornado/web.py", line 1572, in wrapper
  File "/usr/lib/python2.7/site-packages/motioneye/handlers.py", line 208, in get
  File "/usr/lib/python2.7/site-packages/motioneye/handlers.py", line 157, in wrapper
  File "/usr/lib/python2.7/site-packages/motioneye/handlers.py", line 258, in get_config
  File "/usr/lib/python2.7/site-packages/motioneye/config.py", line 1240, in motion_camera_dict_to_ui
  File "/usr/lib/python2.7/shlex.py", line 279, in split
  File "/usr/lib/python2.7/shlex.py", line 269, in next
  File "/usr/lib/python2.7/shlex.py", line 96, in get_token
  File "/usr/lib/python2.7/shlex.py", line 172, in read_token
ValueError: No closing quotation
2015-12-20 20:08:04: [motioneye]    ERROR: 500 GET /config/1/get/?force=true&_=1450638531615&_username=admin&_signature=** (192.168.88.249) 316.17ms
2015-12-20 20:08:04: [motioneye]    ERROR: No closing quotation
Traceback (most recent call last):
  File "home/ccrisan/motioneyeos/output/raspberrypi/target/usr/lib/python2.7/site-packages/tornado/web.py", line 1369, in _stack_context_handle_exception
  File "home/ccrisan/motioneyeos/output/raspberrypi/target/usr/lib/python2.7/site-packages/tornado/web.py", line 1572, in wrapper
  File "/usr/lib/python2.7/site-packages/motioneye/handlers.py", line 211, in get
  File "/usr/lib/python2.7/site-packages/motioneye/handlers.py", line 157, in wrapper
  File "/usr/lib/python2.7/site-packages/motioneye/handlers.py", line 632, in list
  File "/usr/lib/python2.7/site-packages/motioneye/config.py", line 1240, in motion_camera_dict_to_ui
  File "/usr/lib/python2.7/shlex.py", line 279, in split
  File "/usr/lib/python2.7/shlex.py", line 269, in next
  File "/usr/lib/python2.7/shlex.py", line 96, in get_token
  File "/usr/lib/python2.7/shlex.py", line 172, in read_token
ValueError: No closing quotation
2015-12-20 20:08:04: [motioneye]    ERROR: 500 GET /config/list/?_=1450638531613&_username=admin&_signature=** (192.168.88.249) 164.32ms

ccrisan, 2015-12-21:
Could you please backup your configuration, rewrite the image from scratch and restore the configuration? The error is very peculiar and I suspect a disk/partition corruption.

fnordson, 2015-12-21:
This is what I've done after it happened the first time. 
You suggest I try a different SD card?

ccrisan, 2015-12-21:
I suggest rewriting the OS image on either the same SD card or on another one, to see if the problem is reproducible or it's just a partition corruption problem.

fnordson, 2015-12-21:
OK I've rendone this on a different SD card. Same problem.

BUT I tried a different smtp account and it worked without a problem. 
I guess it comes down to the ";" in my password. It's the only special character there and it would explain the errors above.

Solved for me but you should consider this ;)

ccrisan, 2015-12-22:
Hmm I never thought of that. Indeed, the ";" is an important separator of commands to be executed upon various events in motion.conf/thread-*.conf. I'll try to escape it and make sure it's accepted as password. Thanks for reporting the issue and for finding out that the root cause of the problem is.

fnordson, 2015-12-22:
You're welcome.
Thank you for your great work here :)

closed 2015-12-22 by fnordson