motioneye-project/motioneyeos #76: Feature Request - Integrate temperatur sensor

Maschine76, 2015-12-22:
Please integrate temperatur sensor in raspberry pi... temperatur in text overlay...

thx

breakthecode, 2015-12-22:
Oh yes, I´m looking forward to be able to connect a weather station to show information. :+1:

68hc11, 2015-12-23:
Digging deeper into this idea... I think many people could come to this & say "I would like to add X data stream" to text overlay.  How does Mr. Crisan best support this?  Some sort of app/plug in that users could add themselves or does he do all the work himself?  I too can see value of adding a one wire temp probe and passing the data but ultimately I would like to take it from motioneyeos and pass it to openhab or similar automation.

Samfox2, 2015-12-27:
You can do it with a small bash script like this:

#add here your command to get temperature
TEMP=......

#start the things
TEXT="%Y-%m-%d\n%H:%M\nTemperature: $TEMP C"

#Set on Camera Image0
wget -q --delete-after "http://localhost:8080/0/config/set?text_right=$TEXT"

Maschine76, 2015-12-27:
I´ll try it...
many thx

cooleyandy, 2017-01-11:
I SSHed into the motioneyeos RPi and tested the above command.
wget -q --delete-after "http://localhost:8080/0/config/set?text_right=TEST"

It didn't work.
I couldn't get the custom text of the overlay to be set to TEST.
I had the same idea of including temperature data onto the overlay.

bortek, 2017-02-23:
@ccrisan How hard it this to implement OSD functionality that is requested in #76 alternatively add support for temp sensor. I guess the former is easier then people can add their own things not just temp sensor.

ccrisan, 2017-02-23:
@bortek OSD in its proper sense (i.e. overlaid on top of the actual stream) can't be implemented in motionEye; it's done by motion and it's mostly static.

What motionEye can do is [monitoring commands](https://github.com/ccrisan/motioneye/wiki/Monitoring-Commands). On a RPi, you can enable some monitoring stuff out-of-the-box, but anyone can customize it.

bortek, 2017-02-23:
I guess that answer the question. :)

closed 2017-02-23 by bortek

GSVNoFixedabode, 2017-06-24:
@ccrisan I know this is closed but the original question seems to be outstanding: can the Text Overlay be customised with a variable to be saved on the image? Monitoring command work well for OSD but they aren't saved to the static images.

ccrisan, 2017-06-25:
I guess in theory this could be achieved by sending web requests to the running motion daemon.

timdonovanuk, 2018-01-05:
Motion runs on 7999 by default in MotioneyeOS. Having said that, I still cannot get curl http://localhost:7999/1/config/set?text_left=foobar to work.

timdonovanuk, 2018-01-05:
So webcontrol_parms is set off by default in /etc/motion/motion.conf so that needs to be set to 1. However webcontrol seems to be broken in motion 4.x. However I've found another way to set text:

1) Your camera will have a file such as /etc/motioneye/thread-1.conf
2) Create a script that updates this file to text_left or text_right with whatever
3) Send a SIGHUP terminal to motion - this will cause it to re-read the configuration files. To do this, something like:

`sudo kill -s SIGHUP $(pgrep -f /usr/bin/motion)`

ta-da you have updated text on your camera and snapshots. What an over complicated process...

Of course you need to do this as sudo there is the fun and games of making this work in a script too.

timdonovanuk, 2018-01-05:
Or...fixed via https://github.com/Motion-Project/motion/issues/606 thanks again @ccrisan

jasaw, 2018-01-12:
@ccrisan Have you address the omission of `webcontrol_parms` in MotionEye? Sorry I haven't been keeping track on the progress of this issue.

ccrisan, 2018-01-12:
@jasaw not yet but it's one of the priorities.

ccrisan, 2018-02-06:
> @ccrisan Have you address the omission of webcontrol_parms in MotionEye? Sorry I haven't been keeping track on the progress of this issue.

https://github.com/ccrisan/motioneye/commit/197c848e7427c3b0ba901906798a20a972fdaa10

penguinjeff, 2020-04-29:
Oooh just installed motioneyes today and was inquiring google about sensor hookups to motioneyes and ran across this thread. It would be so cool to replace my outdated APC monitors at work with a much easier to use and more up to date interface. It has my head spinning with ideas. When the spinning stops and the ideas coagulate I'll look into all this with renewed interest. I'd like to log temps and humidity and I have ways to do that. I could even setup alarms to email. This motioneyes is good for the cam reqs I'll see how hard it is to add other items to its display. Although I could setup my own alarms I'll check out motioneyes if there would be a way to trigger an alarm with overheating or humidity or battery situation.

starbasessd, 2020-04-29:
On all my other Pis I retrieve and report my CPU temp with the following
command:  vcgencmd measure_temp
I have it send it to me the following script (needs sendemail program to
email it to me)

#/bin/sh
#sleep 90 # pauses 90 seconds for network to come up if using at the bootup
HOME=$(hostname -f)
ip -o -4 route show to default |  ETH=$5
IPADDRESS=$(/sbin/ifconfig $ETH | sed -n 's/.*inet \([^ ]*\).*/\1/p')
GPU_TEMP="gpu_"$(/opt/vc/bin/vcgencmd measure_temp)
TEMP=$(cat /sys/class/thermal/thermal_zone0/temp)
CPU_TEMP="cpu_temp="$((TEMP/1000))"'C"
# echo $GPU_TEMP
# echo $CPU_TEMP
sendemail -f [from email address] -t [to email address] -s [email server]
-u "IP Address" -m "Current Address of ${HOME} is
${IPADDRESS}.\n${CPU_TEMP}\n${GPU_TEMP}" -o tls=yes -xu [sender account]
-xp [gmail password]
I use a BMP280 module to do similar for room temp and air pressure on other
PiZeroWs throughout the house.
the vcgencmd can be used with your own scripts in motioneye for overlays.
Google is your friend.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?