motioneye-project/motioneyeos #77: Only 1 camera is reachable form internet

morbak, 2015-12-22:
Hi,

I've installed MotionEyeOs with 1 RaspiCam on a RsPi. 
IP Lan : 192.168.0.202
I've another RsPi with USB Webcam (jessie + motion streaming)
IP Lan : 192.168.0.201

On the MotionEyeOs I added the 2nd camera (Simple MJPEG camera with local IP : http://192.168.0.201), and i can see all them in a browser.
To acces from WWW, I've redirected the port 80 for IP:192.168.0.202 on another port on my router (Static IP)., so i can reach the Webserver, but i only see the RsPi Camera, the 2nd isn't shown.

I wonder how to show the 2nd camera too, ?

thx,

ccrisan, 2015-12-22:
Forward port 81 to .201? And by the way, never forward 80. Use 8080 instead or some other port. 80 is usually the target of various unwanted requests (attacks and whatnot).

morbak, 2015-12-22:
Ok, it's right the port 80 is often listen. 
But i don't really understand how My 1st RsPi with MotionEyeOs work....this board with MotionEyeOS know the camera2(usb) IP (192.168.0.201), streaming is on port 8081 in my LAN. But when i connect myself on the "webserver" throught WWW MotionEyeOs isn't able to catch the camera streaming to show on the webinterface? i don't understand how to forward "81 to .201".

ccrisan, 2015-12-22:
Nevermind the streaming port 8081; it's not used at all in this scenario. Forward port 81 to your first internal IP and 82 to the second one.

morbak, 2015-12-23:
You mean each IP camera must be reachable from outside with my router static IP through port 81,82,83,.....

ccrisan, 2015-12-23:
Well yes. Or, you can add one of your motionEye cams to the other (as a Remote motionEye Camera) and you could see them both from the UI of the latter. Then you would only need to forward access to one internal IP.

morbak, 2015-12-23:
Ok, but only a motionEye cams, if i've an usb cam without motioneye how to do?

ccrisan, 2015-12-23:
You add it to motionEye :) We're talking about motionEye(OS) here after all.

morbak, 2015-12-23:
Thx I've found my pb the 2nd camera was added as "Simple MJPEG Camera", but if i add as "Network Camera"  it's working.
What are the difference between ?

ccrisan, 2015-12-23:
Simple MJPEG Camera is just a simple frontend for visualizing the MJPEG stream. It never passes through the motionEye server (or motion daemon, for that matter). Therefore, your browser must be able to access the MJPEG streaming URL, which is usually not true when you're outside of your LAN. The advantage however is that you can see the stream without any performance penalties due to server-side processing.

Adding it as a network camera is normally what you wish, taking advantage of everything that motionEye has to offer.

closed 2015-12-23 by ccrisan