motioneye-project/motioneyeos #79: Using both LAN and Wireless

therealwave, 2015-12-23:
I cant seem to get both ports working.
My setup:
I have a Pi2 with a wireless card. I have that working fine. I can connect and configure motionpie on wifi. I have a network (ethernet) ip camera I want to plug into the Pi2 using ethernet port.

If I am doing everything on ethernet using a switch and no wireless it all works fine with the camera.

I have my camera set to 192.168.1.200
Motionpie is 192.168.1.121

Is there a way to get Pi2 wireless connected (192.168.1.121) and have the Motionpie connect to my camera ethernet (192.168.1.200) at the same time?
Thanks,
Brian

ccrisan, 2015-12-23:
motionPie/motionEyeOS was not designed to work with two internet connections at the same time. The only way to obtain what you need is to connect both your PI and your netcam(s) to your local router.

closed 2015-12-23 by ccrisan

drewzh, 2016-03-15:
@therealwave Did you manage to find a solution? I'm in the same boat as you, exactly the same requirement... thinking of possible solutions before jumping over to raspbian/motionEye. The motionEyeOS won't even join the wireless network if ethernet is attached on boot.

ccrisan, 2016-03-15:
It seems to be a simple and clean solution.

njordan77, 2019-07-21:
hm, i have a related issue but maybe not exactly. I started like others i guess via LAN on a Raspi 3A+ USB ehternet adapter, works fine, IP address and all is setup. Now i enabled Wifi and as well receive an additional IP address which is accessable if LAN is still online. As soon as i boot with LAN removed I cannot connect to WIFI port anymore..... ANY idea what is wrong....I remember that i did it the same way lost time but it worked great, this time not....same router, nothing changed.