motioneye-project/motioneyeos #80: Feature Request - Enable / Disable Motion detection through special url

morbak, 2015-12-23:
Hi,

Is it possible to add Enable & Disable motion detection function for all cameras on same time (pre-configured) via an url acces through frontend webserver?

The aim is to drive this function with smartphone (android domotique software, i've already wrote).
When we leave and return back Home... a simple clik on the button will ensure the motion survey

The smartphone application will send to router IP SATIC an url with a special syntax command (Enable/Disable Motion) and with several aleatory parameters to ensure security acces (ie: based on the serial board serail number and a pre défined user code....).

ccrisan, 2015-12-23:
Not yet, but planned.

ozett, 2015-12-28:
if you go further into homeautomation/domotique solutions, you will find (i.e. on esp8266/EspEasy) solutions like url-trigger and also most often nowadays the same with MQTT subscriptions. if you dont find details with google, let me know. but maybe it url-hooks are planned, please think over MQTT cmd at the same time. should be easy to program both with one api...

ccrisan, 2015-12-30:
URL hooks and other API functions are planned indeed. MQTT on the other hand is not on the table yet.

phibes4000, 2016-01-05:
On a similar front, I want to start/stop continuous recording, probably via a WS call implemented in python.  I’m happy to hack up a solution in the meantime - can you please point me to where the ‘movie’ control is implemented?  Can it be changed at runtime (i.e. without restarting processes)?