motioneye-project/motioneyeos #81: Notification emails stop attaching images after approx a week of uptime.

SuperSharpShooter, 2015-12-23:
I'm running ver 20151206 on a RPi 2, and after approx a week of uptime, images stop coming through as attachments with the motion notification emails, but the emails themselves keep coming through. The images and video are still saved locally to the SD card and I can browse to them. I noticed this behaviour on the previous version also. I'm running the RPi cam.

Pulling the power to the RPi and rebooting fixes the issue, but it comes back after approx a week.

Looking in the log, when the empty emails are sent there is a line: [sendmail]    ERROR: timeout waiting for the media listing process to finish. 

Here's what the end of my motioneye.log looks like. These errors only occur in the last day:

2015-12-22 11:29:53: [sendmail]     INFO: email sent
2015-12-22 11:30:28: [sendmail]     INFO: email sent
2015-12-22 11:33:57: [sendmail]     INFO: email sent
2015-12-22 11:34:34: [sendmail]     INFO: email sent
2015-12-22 11:35:20: [sendmail]     INFO: email sent
2015-12-22 11:35:54: [sendmail]     INFO: email sent
2015-12-22 11:36:31: [sendmail]     INFO: email sent
2015-12-22 12:54:17: [sendmail]     INFO: email sent
2015-12-22 13:56:22: [sendmail]     INFO: email sent
2015-12-22 13:56:50: [sendmail]     INFO: email sent
2015-12-22 14:14:41: [sendmail]     INFO: email sent
2015-12-22 14:15:26: [sendmail]     INFO: email sent
2015-12-22 14:41:19: [sendmail]     INFO: email sent
2015-12-22 14:41:46: [sendmail]     INFO: email sent
2015-12-22 15:37:14: [sendmail]     INFO: email sent
2015-12-22 15:37:29: [sendmail]     INFO: email sent
2015-12-22 15:52:33: [sendmail]     INFO: email sent
2015-12-22 15:53:10: [sendmail]     INFO: email sent
2015-12-22 16:00:10: [sendmail]     INFO: email sent
2015-12-22 16:00:21: [sendmail]     INFO: email sent
2015-12-22 19:26:02: [sendmail]     INFO: email sent
2015-12-23 10:17:43: [sendmail]     INFO: email sent
2015-12-23 10:18:18: [sendmail]     INFO: email sent
2015-12-23 10:52:48: [sendmail]     INFO: email sent
2015-12-23 11:21:04: [sendmail]    ERROR: timeout waiting for the media listing process to finish
2015-12-23 11:21:10: [sendmail]     INFO: email sent
2015-12-23 11:21:21: [sendmail]     INFO: email sent
2015-12-23 11:49:55: [sendmail]    ERROR: timeout waiting for the media listing process to finish
2015-12-23 11:50:13: [sendmail]    ERROR: timeout waiting for the media listing process to finish
2015-12-23 11:50:30: [sendmail]     INFO: email sent
2015-12-23 11:50:37: [sendmail]     INFO: email sent
2015-12-23 11:52:44: [sendmail]     INFO: email sent
2015-12-23 13:41:21: [sendmail]    ERROR: timeout waiting for the media listing process to finish
2015-12-23 13:41:26: [sendmail]     INFO: email sent
2015-12-23 13:41:34: [sendmail]     INFO: email sent
2015-12-23 14:33:52: [sendmail]    ERROR: timeout waiting for the media listing process to finish
2015-12-23 14:33:57: [sendmail]     INFO: email sent
2015-12-23 14:35:23: [sendmail]     INFO: email sent
2015-12-23 14:51:07: [sendmail]     INFO: email sent
2015-12-23 16:42:04: [sendmail]    ERROR: timeout waiting for the media listing process to finish
2015-12-23 16:42:09: [sendmail]     INFO: email sent
2015-12-23 17:21:22: [sendmail]    ERROR: timeout waiting for the media listing process to finish
2015-12-23 17:21:27: [sendmail]     INFO: email sent
2015-12-23 17:21:59: [sendmail]    ERROR: timeout waiting for the media listing process to finish
2015-12-23 17:22:04: [sendmail]     INFO: email sent
2015-12-23 17:39:36: [sendmail]     INFO: email sent
2015-12-23 17:49:55: [sendmail]    ERROR: timeout waiting for the media listing process to finish
2015-12-23 17:50:00: [sendmail]     INFO: email sent
2015-12-23 17:51:02: [sendmail]    ERROR: timeout waiting for the media listing process to finish
2015-12-23 17:51:07: [sendmail]     INFO: email sent
2015-12-23 18:19:34: [sendmail]    ERROR: timeout waiting for the media listing process to finish
2015-12-23 18:19:39: [sendmail]     INFO: email sent
2015-12-23 19:08:50: [sendmail]    ERROR: timeout waiting for the media listing process to finish
2015-12-23 19:08:55: [sendmail]     INFO: email sent
2015-12-23 19:14:21: [sendmail]    ERROR: timeout waiting for the media listing process to finish
2015-12-23 19:14:26: [sendmail]     INFO: email sent
2015-12-23 19:14:40: [sendmail]    ERROR: timeout waiting for the media listing process to finish
2015-12-23 19:14:44: [sendmail]     INFO: email sent
2015-12-23 21:43:27: [sendmail]    ERROR: timeout waiting for the media listing process to finish
2015-12-23 21:43:32: [sendmail]     INFO: email sent

ccrisan, 2015-12-23:
Cleanup your media files (reduce the number of pictures/movies taken). motionEye cannot cope with a large number of media files within a reasonable amount of time.

closed 2015-12-23 by ccrisan

SuperSharpShooter, 2015-12-23:
How many is too many images, and what would you say a reasonable amount of time is? Why does it work fine for a week before it fails? It's only trying to send 10-15 images upon triggering once or twice an hour.

Is there a way to adjust the timeout of the media listing process, or to get the process to run less frequently?

ccrisan, 2015-12-23:
You cannot adjust the timeout; it's hardcoded to 10 seconds. The limit depends on the speed of your SD card and many other factors. I recommend keeping no more and a few hundred media files (pictures and movies).