motioneye-project/motioneyeos #84: NFS Mounts

EdGetMeLink, 2015-12-24:
add feature to mount NFS drive

MrJeromeV, 2016-01-11:
Would be very useful to me as I have no Windows computer in my network.

wolfspyre, 2016-01-15:
+1 for this!

pi-pi-pi, 2016-09-20:
I too would love to mount a local NFS share on my LAN. Is there a way this can be done by editing the fstab? I tried that but found the file system to be read only.

Thanks!

Chiny91, 2017-03-16:
Years ago, could be 10 years or more, when I last looked at NFS, it was considered horribly insecure and best avoided, although by now presumably there is suitable authentication and perhaps even encryption.   How about the already included ssh; either sftp with a nice GUI client, or rsync. 

-1 😄 

@pi-pi-pi  Have you tried making the file system writable ?   Info is in the motionEyeOE wiki, under tweaks.

closed 2017-06-01 by bortek