motioneye-project/motioneyeos #85: Can connect to web page in IE but not Chrome

TechGiga, 2015-12-24:
So I put a fresh install of motioneyeos on the sd card on my RPi 2 yesterday, and it was working fine on Chrome. But today I cannot connect to it with Chrome but I can with Internet Explorer. What is up here?

ccrisan, 2015-12-24:
Some screenshots and more details on "won't work" would help :)

TechGiga, 2015-12-24:
Ah, well now that you reply, it works again in Chrome now... It's weird how sometimes it works and sometimes it doesn't. Won't work means it times out.

ccrisan, 2015-12-24:
That's clearly not motionEye related as it makes no difference between browsers. I'll close the issue  now and should you have more details regarding this, feel free to get back to us.

closed 2015-12-24 by ccrisan