motioneye-project/motioneyeos #86: GUI restart

wolverine-forever, 2015-12-26:
I found another issue, what I haven't seen before. After a reboot suddenly the GUI in the browser doesn't want to appear again. I still can use SSH to access the Rasp. Is there a way to restart the GUI so I can see it again in the browser? Until now the only way was reinstalling the Image on the SD card?

ccrisan, 2015-12-30:
You could always restart motionEye using `/etc/init.d/S85motioneye restart`. But that's far from ideal. I would like to see your log files and find the root cause of your problem.

wolverine-forever, 2015-12-31:
Actually I found the reason behind it .. it is option for the network watch. As soon as i switch that off it goes fine. I think it receives a time-out too early .. I will switch it back so I can send you the log maybe you have a smarter way to solve it ..

ccrisan, 2016-01-01:
Yes, please attach all your log files.

Bruzzlee, 2016-01-22:
Hi,
I had a problem that the motioneye service was restarting in e endless loop.
What I've done: Changed the Time Zone --> apply --> reboot --> no Web access.

Login via SSH
motioneye.log:
....
2016-01-22 01:14:51: [motioneye]     INFO: hello! this is motionEye server 20160109
2016-01-22 01:14:51: [motioneye] CRITICAL: media directory "/data/output" does not exist or is not writable
2016-01-22 01:17:47: [motioneye]     INFO: hello! this is motionEye server 20160109
2016-01-22 01:17:47: [motioneye] CRITICAL: media directory "/data/output" does not exist or is not writable

Solution:
mkdir /data/output

2016-01-22 01:20:46: [motioneye]     INFO: hello! this is motionEye server 20160109
2016-01-22 01:20:50: [motioneye]     INFO: cleanup started
2016-01-22 01:20:50: [motioneye]     INFO: wsswitch started
2016-01-22 01:20:50: [motioneye]     INFO: tasks started
2016-01-22 01:20:50: [motioneye]     INFO: mjpg client garbage collector started
2016-01-22 01:20:50: [motioneye]     INFO: smb mounts started
2016-01-22 01:20:51: [motioneye]     INFO: server started

webserver was back on again.
hope this helps.

Bruzzlee

bortek, 2017-02-23:
There was another user that has the similar problem with changing of timezone. His problem was resolved by upgrading to the latest version. Please try to upgrade and reopen the Issue is the problem persists.

closed 2017-02-23 by bortek