motioneye-project/motioneyeos #87: Feature Request: Support for RaspberryPi Zero

endomorphismus, 2015-12-27:
Hi,

Are u planning to cook a build for the raspberry pi zero? I got a USB splitter to run a Wifi dongle and a USB webcam and it works with the motioneye os for the RPi one build but it's flaky. Getting this OS working with the zero will be killer since the zero is so cheap.

Thx!

ccrisan, 2015-12-27:
As a matter of fact I'm planning to release a new version today. What exactly is not working right with Zero? Normally the RPi1 version should be fully compatible with the Zero.

endomorphismus, 2015-12-29:
Running 20151206, added an USB web camera then got a spinning wheel, does spin forever. So i reboot and log in i get again "An error occurred. Refreshing is recommended" and "You have not configured any camera yet. Click here to add one...". When trying to add a camera i don't see the USB webcam anymore (but the light is on on the webcam).

ccrisan, 2015-12-30:
Please attach your log files, @endomorphismus.

bortek, 2017-02-23:
I am using this on Pi B+ and on pi zero with pi camera and it works great. Closing issue.

closed 2017-02-23 by bortek