motioneye-project/motioneyeos #88: establish mesh network

ben19871987, 2015-12-27:
hi dear
in motioneyeOS there is olsr package and it helps me to create wireless ad hoc network but how can i establish and manage mesh network.

yours

ccrisan, 2015-12-30:
I'm sorry but support for ad-hoc networks is not something too many people would use. If you need to set up such a network connection, please use Raspbian with motionEye.

closed 2015-12-30 by ccrisan

acwest, 2020-06-17:
with batman-adv support, this would be really hand to build out a network of pi zero w boards that communicate directly with each otherf. unfortunately 802.11s won't work with that board, but other protocols will. I'm going to see if it is feasible to build an image that works...

starbasessd, 2020-06-17:
A simple PiZero set as a AP/Hub, then have the rest connect as needed to it
is easily doable. True mesh may not be as easy, but toss in a second wifi
dongle on them and bingo, working. All with the current image.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

acwest, 2020-06-17:
I tend to agree that true mesh wouldn't be easy, but I am looking at a common security camera problem, a large number of cameras scattered over a fairly wide area, only some of which would be able to reach a hub. In my case, they are surrounding a house, and the walls block wifi on some of the sides of the building, and there is no one point that is reachable from everywhere. The same thing would also work really well for wildlife cameras in a forest (power is an issue, but solvable, those cameras already exist, after all)
I am currently looking at building from source, then adding the correct bits to see if it works

starbasessd, 2020-06-17:
Cool beans. I watch eagerly, as an alternative.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?