motioneye-project/motioneyeos #89: Feature Request: mask for none used areas

wolverine-forever, 2015-12-28:
Would it be possible to build a mask so for example a tree would not be calculated in the 100%.

Let me explain the reason. I have a garden where i have several trees ... when there is wind i need to increase the % change to cover the change made by the tree. However 10 minutes later it stops blowing and the tree not contributes. This could occur in that some one could pass by on some distance without that it suddenly will detect the change due to the higher %. If I could black out the tree (where i do not expect to come a treat from) I would not need to compensate for the wind and could keep a low % in the change to detect motion in a proper way.

Let me hear your toughts about it ....

By the way thanks for a great piece of software ..

ektus, 2015-12-28:
You're not the first with that wish. The good news: It is planned, and as it is already there in motion (the underlying detection software), it's even possible to use it now by providing a mask image and adding the appropriate command to the "Extra Motion Commands" section in the configuration of the Video Device.

https://github.com/ccrisan/motioneye/issues/15
https://github.com/ccrisan/motioneye/issues/50
http://www.lavrsen.dk/foswiki/bin/view/Motion/ConfigOptionMaskFile

Regards
Ektus.

wolverine-forever, 2015-12-31:
He Ektus,

Where is the file located? I tried to find the motion.conf file in the way described by Lavrsen, but couldn't find the tweak.

Regards, Wolverine_forever

ektus, 2015-12-31:
You must create a mask.pgm file with the right size as described by Lavrsen and store it somewhere in the data partition. Then, simply add the command in the motioneye GUI as described earlier. I'm using motioneye under raspbian, and my command reads "mask_file /home/pi/motion-mrdave/mask.pgm"

Regards, and Happy New Year
Ektus.

wolverine-forever, 2016-01-01:
Hi Ektus,

First off all best wishes for the new year and thanks for your comments.
How to check if the mask is applied?

Regards,

ektus, 2016-01-01:
Mask off half of the field of view, optionally set "Show Frame Changes" to active and move around in front of the camera? It should record images only if the motion is in the area that is not masked.

Regards,
Ektus.

Chiny91, 2016-01-03:
Alternatively, the log files can be checked for *mask* entries.

I find the level of normal logging does not show a success (only problems) for loading the mask file, so I *Enable debugging* in *Expert mode*.   After the required reboot, clicking on *messages.log* will contain a line like:
*user.info motion-mrdave: [3] [INF] [ALL] motion_init: Maskfile "path_to/mask_filename.pgm" loaded*
The debug log is best clicked off (another re-boot) after the check.

Devising a useful mask file is quite an art; worth a read of the details:
*http://www.lavrsen.dk/foswiki/bin/view/Motion/ConfigOptionMaskFile*

wolverine-forever, 2016-01-03:
Chiny91,

Thanks a lot ... that is really help full and also showed me that the mask was not working ... after another try it was loaded as it should be ... :) Now i could optimize the Motioneye settings so it catches the real things to catch

Hexagon, 2016-02-10:
I too would like to see a built in mask editor! Too increase the chances i took the liberty of making my owm pgm reader/writer, and a javascript editor that wouldn't be too hard to implement :)

https://github.com/Hexagon/maskitor

A working example of the editor is available at http://hexagon.github.io/Maskitor/example/

Please note that this is in an early stage of development, so I wouldn't say it has a lot of features. To add masked area, left click at a "pixel", to remove mask, right click at a pixel. To fill a rectangle, hold shift while clicking and draggin with left mouse button. To clear a rectangle, do the same thing but with right mouse button.

To implement this, one would have to include maskitor.js in HTML, and add javascript to get a recent frame, load existing mask, and to save the new mask. ```example/index.html``` is well documented, but I'll include a quickguide below.

```javascript
// Initialize editor, parameters:
//   1. Id of container element in where to place the editor
//   2. editor width
//   3. editor height
//   4. camera resoltion x
//   5. camera resoltion y
var editor = new Maskitor("container", 640, 480, 1280, 960);

// To set backdrop (current camera view), pass a js Image-object to setBackdrop
editor.setBackdrop(backdrop);

// To resize editor
editor.resize(newWidth, newHeight);

// To load current mask, read the file in binary format, and pass that into setMask as a Uint8Array
editor.setMask(new Uint8Array(rawFile.response));

// To get gpm-file from editor, create a new PGMFile
var pgmFile = new Maskitor.PGMFile(),

// Populate pgmFile with data from maskContext, save the binary file representation (Arraybuffer) in variable binaryData
binaryData = pgmFile.fromContext(editor.maskContext).toBinaryData();

// Send binaryData to server, and save it to the destination file
```

I'll continue to develop this when i find time, but the api should'nt have to change much, if you choose to implement it.

Anyways, thanks for a great piece of software. I've been looking for something that just works for quite a while :)

cottaer, 2016-02-28:
Hello, I have a Raspi 2, remotely connected with 3 ip cameras
Extra motion option (mask file) works exactly with only one camera, with the last configured camera. the thread-x.conf of the remote ip cameras are identical.
If I configure a second camera with extra motion option (mask file) with the GUI stops motion detection on the 1. camera completely, the same at the next camera .....
It works motion detection with extra motion option (mask file) only at the last configured camera.
no extra motion option (mask file) works motion detection well, but produces too many pictures.
CurrentVersion 20160221, the same with the previous version
Thanks for ideas, thanks for moioneyeos

ccrisan, 2016-10-04:
Added in https://github.com/ccrisan/motioneyeos/releases/tag/20160919.

closed 2016-10-04 by ccrisan