motioneye-project/motioneyeos #90: Feature Request: Overlay-Icon for direct recording/screenshot

ozett, 2015-12-28:
action-buttons are wonderful since 0.30, but icons/buttons for some direct screenshot/recording  would  be wonderful too. i am used to this from other webcam solutions. 

By the way thanks for a great piece of software ..

ccrisan, 2015-12-30:
The two action buttons are already there as a matter of fact. They are hidden however as I haven't yet figured out some issues with their actual implementation. To be more precise, interacting with the motion daemon is not always piece of cake.

Samfox2, 2016-01-01:
It would be great to have a user defined action button that could start a script - I would use it to turn on my door light.

ccrisan, 2016-01-01:
@Samfox2 it's already there as of 20151230: https://github.com/ccrisan/motioneyeos/wiki/Action-Buttons.

Samfox2, 2016-01-02:
I don't know how I missed that. Tried it and It is working, thanks!

closed 2017-02-23 by bortek