motioneye-project/motioneyeos #91: Not able to get Raspberry Pi NOIR camera to work with motioneyeOS

Crazyslade, 2015-12-29:
Hello, I have been all over the internet and this thread. I have not been able to find a solution as of yet.  I have a Raspberry Pi 2 with motioneyeOS installed on a 16g micro SD card.  It boots up great and I can get logged in.  I go to add camera and select the camera that it sees (mmal service 16.1) and add it.  After it finishes adding it there is a preview window that is grey and says camera1 up in the upper left hand corner and a No Camera symbol in the middle. It acts like its reading it but no stream.  I have had this camera working in the Raspberry Pi B but it does not fit in the camera housing I am using.  I am real new to the linux scene so try not to laugh to hard at the questions you will probably get from me. Thanks for any help you might be able to give.

Crazyslade, 2015-12-30:
This is what i am getting 
[image]

olli84, 2015-12-30:
Hey there,
i got exactly the same issue. Same Hardware, NOIR Cam and no picture in the preview window. I had to say that i use Fast Network Camera only and use Zoneminder as frontend for my cameras. The funny thing is - in Zoneminder everything is great. I can watch my stream without any purpose. I checked twice using Safari and Chrome. Reboot -> same issue.

Thanks for your help!

ccrisan, 2015-12-30:
So does it work as fast network camera or not? Have you enabled streaming authentication?

olli84, 2015-12-30:
Thanks for your fast answer. It works, but i cannot see anything in motioneyeos itself. There is only the camera symbol like in the picture in the second post. I need to check in Zoneminder to watch my stream. Basic Authentication is on.

ccrisan, 2015-12-30:
Please try disabling the authentication and see if it works then. There's is (was) a known problem with streaming authentication and motionEye's preview, although it shouldn't affect the fast network camera preview.

Crazyslade, 2015-12-30:
Hi ccrisan, I have never enabled authentication and I just tried Enabling Fast Network Camera. It changed my camera to an Simple MJPEG Camera but still no preview and the firefox browser just continues to load. Thanks for helping us work on this!

olli84, 2015-12-30:
Hey there,
disabled authentication doesn't help. It works always with my pi1 b+. Don't know what happened now. Embedded url and stream url gives no picture. I checked again with zoneminder, live picture is black, motion detection and recording works (what the hell is going on?). I'm able to watch recorded events.

ccrisan, 2015-12-30:
@olli84 please attach your log files.

olli84, 2015-12-30:
Your response time is insane. Thanks a lot. I disabled zoneminder for this cam to filter the problem.

I can give you remote access to that camera if you want.

[Archiv.zip](https://github.com/ccrisan/motioneyeos/files/75076/Archiv.zip)

ccrisan, 2015-12-30:
There's nothing suspicious in your log files, (un)fortunately. Would you be able to give me SSH remote access?

Crazyslade, 2015-12-30:
I have returned my camera and now waiting on a new one to see if it was the camera that was defective or a victim of the dreaded static attack.  I will update in a few days so that we can continue or close as resolved.  thanks guys

olli84, 2015-12-30:
Hey Calin,

i've sent you an e-mail with the access for my cam.

Again, thanks for your work.

Crazyslade, 2016-01-04:
I got my new camera in today and it works with motioneye. So as far as my part in this goes it has been resolved. Thank you for your help.

closed 2016-01-04 by ccrisan

superddr, 2017-09-23:
@Crazyslade when you said you got your new camera, you meant you bought a new USB camera instead of a new rpi camera?

Crazyslade, 2017-09-25:
@Superddr I just got the same too camera. I assume that I had static issues with the first one.

Crazyslade, 2017-09-25:
Not "too" but "rpi"