motioneye-project/motioneyeos #92: names and definitions change/foscam fi9903p motion deteciton triggert/smart_mask not working

885narrow, 2015-12-29:
Dear Crisan,
being still in the testing phase for two months now, I would like to thank you for the time you spent on making such a nice project.
I am still trying to figure out some issues I have with my setup: 

sometimes the camera names and of course definitions like file storage etc. change and after some time they are back to what they were.
I use USB camera c920 logitech, the raspicam and the foscam fi9903p (rtsp).

The issue with the rtsp and the foscam is maybe because too many pixels are changing on delivery of the stream from the camera. It is permanently recording even there is no motion.
In setting the show frame changes I can see that often the whole picture is taken.

What is most important for me is that smart_mask seems not working. I installed motion on the raspbian and smart_mask it is working fine , all the trees in my view do not trigger motion. Using motioneyeos and the permanent setting for smart_mask does not have the same effect. Moving trees are causing a motion alarm all the time.

Many thanks again for your work and the opportunity to ask for advice here.
My Best Regards
Roman

Chiny91, 2015-12-29:
I've also been trying the mask feature and have the same problem.   I have checked the obvious gremlins (mask in captured picture to be same size, mask readable by all, mask is pnm within .pgm file).   No joy.

/data/log/messages has two lines that look relevant:
Dec 29 19:35:44 meye-2c2ea82e user.err motion-mrdave: [2] [ERR] [ALL] get_pgm: Failed reading maximum value in pgm file: Operation now in progress
Dec 29 19:35:44 meye-2c2ea82e user.err motion-mrdave: [2] [ERR] [ALL] motion_init: Failed to read mask image. Mask feature disabled.

So, my problem is that *Failed reading maximum value in pgm file*.   Thinking...

Chiny91, 2016-01-03:
It turns out that *motion-mrdave* is especially fussy with the format of .pgm files.   I imported my failed mask into a different pixel editor and re-exported it.   Success; I now have a working mask file.

The software I successfully use to produce .pgm files is *Gimp*, a multi-platform freebie.   .pgm export options are *pnm* and *raw*.

motioneyeosuser, 2016-03-12:
Hello, i want a logitech C920 operate on motioneyeos.
Unfortunately I get a gray box with the message " Unable to open Video Device ".
The power supply from the raspberry has 2 ampere . I need an active usb hub or if another problem before ?
a simple logitech usb camera works fine .

885narrow, 2016-03-12:
Change the resolution video on the c920 to 1280x720 ..

885narrow, 2016-03-12:
I didn't get smart_mask to work so I am using motion on ubuntu straight now.....

bortek, 2017-02-23:
Is this still an Issue?

885narrow, 2017-02-23:
Dear Bortek,Thank you for asking, but I don't know since I stopped using it due to this problems.I use motion on ubuntu which works fine for me.All the best,Roman

closed 2017-02-23 by bortek