motioneye-project/motioneyeos #93: Start/Stop motion by ssh

Mikiya83, 2015-12-29:
Hi
I need to control many cameras from one "authentication server" point. My other camera is installed on a Debian system, and a standard Motion run on it. So i can start/stop Motion from the server by a SSH script.
But i can't find a way to do that on motionEye OS with a RPi. Your motion process is not started with a service, how can i start or stop it from a server (ideally by SSH) ?

Thanks !

Mikiya83, 2015-12-30:
I answer to myself : if the process "motioneye" is stopped or started by the init.d script, the motion process is stopped/started too. It's not a perfect solution because all is shutdown in that case, but it's enough for me ;)

ccrisan, 2015-12-30:
I know people are asking more and more for the possibility to control themselves the motion process, the motion detection mechanism and other related functionalities. I have planned to implement an API, a way to do all these things, but I haven't gotten to do anything about it yet. It's good that people manage to find various workarounds until I actually get to implement this promised API.

SirMarco, 2016-03-20:
@Mikiya83 
do you use SSH with password protection ? I would like start and stop motioneyeos with my homecontrol (FHEM)

dimgod59, 2019-07-26:
> I answer to myself : if the process "motioneye" is stopped or started by the init.d script, the motion process is stopped/started too. It's not a perfect solution because all is shutdown in that case, but it's enough for me ;)

hi, can you explain your method in the "terminal" ? ssh motioneye service stop ?