motioneye-project/motioneyeos #94: Feature Request Avahi

steveww, 2015-12-30:
Please include the avahi-daemon in the OS image. This would make finding the device on the network much easier via mDNS.

ccrisan, 2015-12-30:
I'd have to read a bit about that as I'm not really sure how it works, but yes, normally I don't see why not. Unless of course it pulls a large amount of dependencies which would break the *lightweight* nature of motionEyeOS.

steveww, 2015-12-30:
It depends on dbus for client notification, so not too bad.
Avahi could then be used to advertise the web address, ssh and possibly other services like the video stream from the camera. Additionally motioneye could use the client to auto configure additional cameras etc.

bortek, 2017-02-23:
Guys cal you elaborate if this is still important. Thanks!

ciscomonkey, 2017-05-17:
I'd also like to see this added to the default install.  If you need configuration files for the different services let us know.

StewartSnow, 2018-06-04:
I’d love to see this in there. I can see that avahi is already in the packages list but doesn’t appear to be operations all by default. Anyone know how to enable it??

jasaw, 2018-06-08:
@StewartSnow You could try enabling avahi and building motionEyeOS image from source.
Running `./build.sh raspberrypi menuconfig` will bring up the image configuration. Look for Avahi and enable it, save it, and compile.

Other than enabling the package, do you know what configuration is needed to get avahi to work?

StewartSnow, 2018-06-09:
@jasaw I believe by default you don't need any configuration for this package. Once it's installed it will pick up the current hostname and announce it's mdns name to all other clients on the network as [hostname].local.
It's so useful as you no longer need to continually keep track of IP addresses, setup manual DNS records for each host, or set static IP addresses. It just works.... no config needed. Would be great to have available by default - and could pave the way for the interest in camera discovery - as this is basically the same as the Apple Bonjour service - used for discovery of services on a network :)

macsip, 2018-11-30:
Is there any possibility to see this feature implemented? It would be a great improvement.

eretna, 2019-05-29:
Any news about it?

JamesCordell, 2019-08-29:
This would mean I don't have to plug it/ trawl the DHCP sniff the network in to find the IP address. This is a must have  :)

morrisonj, 2019-11-18:
Agreed. This is would be a good feature to have. I've just come looking for it now. Shame it's not possible yet.

zackkitzmiller, 2020-05-25:
Agree. This would be excellent.

starbasessd, 2020-05-25:
IMO waste of time, as the default name is effectively meye-<large random
number>. If it called itself meye-newcamera or meye-newhub or even
motioneyeos it would be like the RPI's default of raspberrypi.local.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

zackkitzmiller, 2020-05-25:
It’s not a waste of time, and hostnames are configurable. When you have
something like 200 devices in your network, and you don’t want to ansible
or chef everyone, service discovery is incredibly important. I use consul
as a routing mesh and I need it to resolve DNS as expected. Even with the
hash it’s still not discoverable.
The best thing I would do for now is add this particular device to my local
DNS server and hope the static IP never somehow gets messed up, or it gets
moved to a different subnet.

In any case. I find mDNS to be incredibly useful for this sort of device.

morrisonj, 2020-05-25:
> IMO waste of time, as the default name is effectively meye-<large random number>. If it called itself meye-newcamera or meye-newhub or even motioneyeos it would be like the RPI's default of raspberrypi.local.

You would set a unique hostname for each Pi to something useful and memorable like meye-frontdoor.local or meye-whatever.local. I would find this very helpful in my install of 8 Pi's running motioneye...

starbasessd, 2020-05-25:
Think about what you said.
      'hostnames are configurable'
They are, but AFTER you remote into it. At least until CCrisan adds
'hostname' to the list of pre-configurable (/boot/...)  files..
Or use the MAC address instead of the serial number, or some other way of
not using a random number.

     When you havesomething like 200 devices in your network, and you don’t
want to
     ansible or chef everyone, service discovery is incredibly important.
Granted. I have about 100 devices on my local home network, myself. 80% are
set up via MAC address static issue IP via my Ubiquity router.

     I use consul as a routing mesh and I need it to resolve DNS as
expected. Even with the hash it’s still not discoverable.
Again, granted.

     The best thing I would do for now is add this particular device to my
local DNS server and hope the static IP
     never somehow gets messed up, or it gets moved to a different subnet
I basically have to fire up a new PI, with RASPBIAN, and get the MAC that
way, if I plan on it being part of my regular network, or
hunt through the DHCP interface on my router and look for a new
meye-<random number> ( in the pool of open DHCP addresses)
I hate doing anything multiple times, myself. .
I think it might be easier to put an option in 'os.conf' or a seperate
file'hostname.conf' that you can put in /boot, like the statc_ip
configuration files, or ssh, or wpa_supplicant.conf files.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2020-05-25:
> IMO waste of time, as the default name is effectively meye-. If it called

Jim:
My point exactly. You currently can't set a name before first boot, the
default name is  'meye-<random number> ' and thus avahi and other 'find by
name services' don't exactly work before you remote in and change the name,
and you can't remote in by name until you change such name, or look at the
DHCP server to find the name or ip address.
I have a number of PI motionEyeOS devices, cameras and hubs, and the first
thing I have to do is go hunt them down.
PITA. At least until I have the device in the Static IP list.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

jasaw, 2020-05-25:
>When you havesomething like 200 devices in your network

In this scenario, you would configure the host name by creating `/boot/hostname`. From memory, I think it is already supported. Can you guys please try it and report back?

starbasessd, 2020-05-26:
@jasaw:
Confirmed.
created plaintext file 'hostname', contents 'rpi3b-meos' with notepad++ on
Win10, placed in /boot/ partition when imaging SDCard.
Worked.
Where is that option documented? It's not in the motionEyeOS WIKI that I
can find...

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

StewartSnow, 2020-05-26:
It could perhaps be worth pointing out that the value of discovery, in my opinion, extends beyond the pure initial setup which is being debated over the past few comments, but also in ongoing access.

If you have a large number of motioneye instances, being able to access them via motioneyehostname.local is very helpful - rather than needing to mess about with DHCP reserved addresses, internal DNS servers, or other means for accessing a large number of devices via easy name. 

Especially since the point mentioned above about being able to pre-define the hostname in the boot config.

jasaw, 2020-05-26:
@StewartSnow Forgive me for my lack of understanding on exactly what Avahi buys us. I've been accessing my motionEyeOS devices via motioneyehostname.local method for years now. From my understanding, when motionEyeOS requests for IP address via DHCP, it also announces its hostname, which the DHCP server stores against the IP address assigned to it. When I send DNS request to my router to resolve motioneyehostname.local, my router resolves it correctly.

If you don't mind, can you please educate us on what Avahi does? Again, I apologize for my lack of understanding.

StewartSnow, 2020-05-27:
Hey @jasaw,
Sure - my understanding (rightly or wrongly) is that avahi or mdns service running on a device enables the announcing of a bonjour based hostname address ie motioneyehostname.local. And that without it, you cannot resolve these addresses - it's certainly the experience that I have noticed wherever I've implemented this approach.

admun, 2020-07-19:
I tried to /boot/hostname method, it does not work.

In fact, the file disappeared in /boot once boot.. not sure why.

It'd be nice to have mdns setup for motioneyeos

starbasessd, 2020-07-19:
Which version of motionEyeOS?
Files placed in /boot usually are deleted once read and / or moved to
proper folder
Are you using Windows or other OS to connect / and / or burning the image?
Did you try <whatever_hostname>.local as well as <whatever_hostname> to
connect by name?
To confirm hostname change, log into the PI with ssh or PuTTY, or look in
the upper left corner. It should have whatever hostname you put in there,
ranter than meye-xxxxxxxx

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

admun, 2020-07-20:
Here's what I am on

motionEye Version	0.42.1
Motion Version	4.2.2+gitUNKNOWN
OS Version	motionEyeOS 20200606

I am on Linux, mounting the /boot and add hostname.

I ssh and saw on the GUI that hostname was changed, but I do not see mDNS working (as reported above)

I tried ssh both with or without .local, but getting host not resolve.

starbasessd, 2020-07-20:
Do you have your local router/dhcp server set as the first DNS server in
DHCP, or on the Linux machine?
IOW if your local router/DHCP server is 192.168.1.1, is your DNS Server
setting set to 192.168.1.1, 4.2.2.4, 8.8.8.8 or only to the external DNS
servers?
My local router handles all internal requests and is listed first.
The command on your Linux machine

nmcli device show <interfacename> | grep IP4.DNS

Will show what DNS server(s) you are using, and in what order.
If your local router/dhcp server isn't #1 normal DNS will never work.
Mine shows:
IP4.DNS[1]:                             192.168.2.1
IP4.DNS[2]:                             8.8.8.8
which is correct for my network.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

admun, 2020-07-21:
I have my router as DNS resolver Google Wifi.

Welcome to meye.local!
[root@meye ~]# cat /etc/resolv.conf 
domain lan
search lan
nameserver 192.168.86.1 <---- this is the gateway of my Google Wifi, which also act has DNS resolver
[root@meye ~]# hostname -a
meye.local

Note that I have other raspberry pi running Raspbian and it has avahi. They are discovered as pi-host.local. So, I know mDNS is working in the local network.

ugoi, 2023-10-04:
Any updates on the progress of avahi-daemon feature?

starbasessd, 2023-10-04:
Non-existent. Little advancement of motionEye, and no advancement of motionEyeOS in 3 years. Avahi was not a feature of buildroot.