motioneye-project/motioneyeos #95: Add a message section like 'Issues' for hardware setups etc?

birukun, 2015-12-31:
I wanted to share my setup with others so that they know what works well and sometimes the casual user may not know what cameras to select, or good configs.  These are not 'issues' so not really for this section.  

Can you add a section for users to contribute their setups and discuss tips etc for MotionEyeOS? 

Example: 

Platform : Pi 2 / 16GB / cabled connection to router
Camera: Logitech C310  set to 1280 x 720
Set up to email notifications, using GoDaddy email account, 15 stills each time.
Other comments: lamp on timer triggered motion detection to confirm continued operation and that lamp was on/off at scheduled times. I received emails when lamp came on and later turned off every day while on holiday.  Can add other parameters to this thread on request.

Worked really well for me on holiday and such a low cost low power solution = happy user! THANKS

ccrisan, 2016-01-01:
Well let's use this issue with the purpose of sharing our configurations and impressions.

Mikiya83, 2016-01-02:
Hi
What about FPS ?
On Pi 1 set on "Modest OC", with Pi camera, i can't even reach 2 fps (set to 640*480) with Motion detection "on", email, and still image recording (without movie, to relieve the CPU). I swith off services except FTP and SSH.
Any advices ? 
Thanks !

Chiny91, 2016-01-02:
Seems a good idea:

Platform: was Pi B, now Pi 2; no real difference noted.   Ethernet.
3 * Hikvision 2032 bullet cameras on 704 * 576, all at 4 fps
Using http, about 23% Pi 2 cpu.   Swapping 1 cam to rtsp goes to 63%
Assorted friends & family have remote access, works well.
Only ssh service running.

Pi B system ran for 2 months unattended, 800k pics, mostly of a family of deer running round my garden.   Current Pi 2 system rather prone to failing to respond to anything but ping (not http, not ssh) after a day or so

I can stream at 10 fps (all 3 cams) easily but have never been able to get more than 1 still per second - config problem somewhere.   I do hack the motion.conf and thread files where no GUI equivalent exists.

After 11 months of use, the SDcard (16 GB Kingston) failed.

*I have been using motion for many years and great to discover such a useful implementation.*

68hc11, 2016-01-03:
I'm trying to develop a personal home security system using motioneye. 

Platform : 2 Pi B+ set on "Modest OC", SD card: Sandisk ultra class 10 
               : 1 Pi 2 set on "Modest OC", SD card Transend class 10
 Camera : all Pi camera 
 Services : Motion detection, email (not working yet) and still image recording when motion triggered (without movie, to relieve the CPU).
 Res. : 1440 x 1024 ea
 FPS : (how do I measure this?)

I am still in the early learning, experimenting phase going on 3 months.  I clearly need a camera that can do good night vision and am interested in the Hikvision series (but how is this integrated into Motioneye OS?).

I clearly need ability to mask vegetation motion due to wind and think there are enough clues here how to do that now but have yet to try.

I haven't been successful at setting my gmail account to mail me yet.

I plan to build a wired cat 6 network with POE to run the 4 remote pies.  (will Gigabit be worth the extra $?).

I'm debating if I should continue with Raspberrys or move to PineA64 in a few months. (to get gigabit & 64 bit)

I "think" I want higher resolutions stills versus movies, since identifying people is my main focus here.

So far I've learned a good bit leveraging MotionEye OS & raspberries I had laying about.  I've proved the concept to my family (we can watch cameras over our phones in the house), but I've yet to trigger emails or set up remote log in monitoring.  Ultimately I'd like to couple this with openhab or similar remote technology for alarms & monitoring.

Thanks!

Chiny91, 2016-01-03:
**68hc11** *interested in the Hikvision series (but how is this integrated into Motioneye OS?)*

I have 3 Hikvision cams type DS-2CD2032-I with PoE running - which one/s do you have in mind ?

Mikiya83, 2016-01-03:
For gmail : 
smtp.gmail.com
port 587
use TLS
your account with gmail.com and password

It should work ;)

For the FPS, it's not that easy to mesure, you can set a FPS max but the FPS in video is not regular ... i use security stream in my NAS app to see it.

During my tests, i found even the Pi B+ CPU is enough to get 640*480 @ 5/6 FPS but if i enable recording in SD card, that drop to 1/2 FPS. The recording is quite hard without a HDD or SSD... if you can get a board with a good CPU (especially if you want 720p @ 7/8 FPS) AND with USB3. or Sata port it should be perfect (for example : Odroid XU4).

68hc11, 2016-01-04:
Chiny91, I would like to purchase a single outdoor (south FL weather), camera to mount under my garage eaves and shoot across my driveway (50ft wide) and out to my mailbox about 120ft out.  .(I probably will go with a varifocus to learn more about field of vision).  Ultimately I would like something to enable "usable" night vision (Is there any camera that can block car headlights enough to ID a car tag?).  Using a 3rd party camera with Motion OS requires the Motion OS "server" to perform the motion detection?  Is there any advantage to some of the features of the Hikvision cameras (triggers etc) and can that be realized in MotionEye OS over POE?  I the future of surveillance is interesting, motion/vector analysis should help ID people vs animals and thermal imaging may soon be reasonable for home usage too.

I like Mikiya83's cheap Linux server (Odroid XU4), that seems like just the thing for the closet recorder.

When using the Hikvision cameras do you also use the native recording as well as streaming or is this not possible?

Chiny91, 2016-01-04:
@68hc11 It is probably off-topic to discuss camera selection here, even if I were an expert.   You may already have seen it; certainly I can recommend https://www.ipcamtalk.com/forum.php who will be able to help.

You can indeed use both the Hikvision system within each camera at the same time as motionEyeOS, although I did so purely as a test.   These days, I use just motionEyeOS, as this brings the cams together in one web page, is more tweakable and I am used to *motion* that I have been using for years on other Linux platforms.  In my experiments last year, I did not find any advantage of the Hikvision system over motionEyeOS but YMMV.

Good luck with your investigations - after all, it is half the fun :smile:

niallie, 2016-01-19:
I'll chip in with my settings, because I've also been struggling with this issue. I've now got it working where motion detection functions and video recording is at 10fps, with a Pi2B and a Pi Camera Module.

Services: All FTP ON | Samba Off | SSH Off
Fast Network Camera: OFF
Video Resolution: 640 x 480
Frame Rate: 10fps
File Storage: USB Key 16GB plus upload to Google Drive
Video Streaming: OFF
Still Images: OFF
Movies:
Quality - 25% | Motion triggered | Minimum length 30 seconds (doesn't work!) | Preserve for One Week
Motion Detection: 1.5% | Auto Noise Detection: OFF | Noise level 12% | Motion Gap 10s | Captured Before 0 | Captured After 100 | Minimum Frames 10
Notifications: OFF
Schedule: Inactive

I think the key thing is to keep the resolution low and try not writing stills at the same time as movies.

olliraa, 2016-02-07:
Hi, Just a few questions on best settings. Running rpi2 and 3 x HikVision ip cameras. Currently the config is like this.

Cameras (camera control panel):

1280 x 720
h264
qyality: medium
max bitrate 4096
fps: 15

Motioneye settings:

movie/camera settings
fps: 15
quality: 50%
motion triggered
1280 x 720

It seems that the poor rpi2 is struggling with these settings pretty bad. CPU load with top is ~80-98 all the time. And the fps overlay shows recording fps 4-8 all the time -> not reaching the target 15 fps.

Is there anything I can do (other thatn lower the resolution)?

closed 2017-02-23 by bortek

ArmandH, 2017-05-18:
What is the best setting for if i want when a motion takes place to record for lets say a half hour?

THORsHand2018, 2018-12-14:
Hello,

MotionEye runs as a virtual machine
Samba on, FTP off, SSH on, Netdata installed
OS - Debian 9.6
CPU - Intel Xeon E5645 6 core@ 2.4GHz, 2 available, 1 allocated
RAM - 24GB available, 6 GB allocated
Storage - dedicated WD Purple 6TB surveillance HDD
Camera - 3x UNV IPC2122LR3-PF40-C 1080p@25FPS
Dedicated network card inside my server, with it's own DHCP server (external) and CCTV Switch
1x Continous recording with 10 minute limit per movie, encoded in H264
1x Motion detection-triggered recording, again H264, 10 minute limit
1x Motion detection-triggered snapshots
all cameras are set to max res, max fps
--------------------------------------------------------------------------------------------------------
CPU Usage doesn't get above 50%, there are 80% spikes when i do something like access the recordings list, or applying settings
RAM usage is low, 0.50GB active, another half-gig as a buffer, the bulk of it is in the cached section.
HDD space is enough for roughly 11 days of keeping recordings
performance isn't ideal, but it's acceptable, personally i'm not satisfied with it, i used motion eye because i like it, and i had a spare server, and it runs on linux, an OS witch i'm intrested in, i get the 1080p resolution, but the quality is worse then the camera provides, in it's own android view app, everything looks crisp AF, but in meye, looks more pixelated then i'd consider 1080p due to something keeping my bitrate low (around 7MB/s for all cameras, that would be around 2.3mbps for each camera, 1080p 25FPS... that;s a low bandwidth, still looking for ways to force it around 10mbps, to have enough bandwidth for highest quality and framerate the cameras can provide. if i could manage to make it happen, meye would get 50% recomandation from me, it's a very neat thing, you can learn a lot of coding by fiddling around (and a lot of networking if you keep digging ), but at the same time it's way of doing it's job is highly inefficient due to the motion deamon mostly.