motioneye-project/motioneyeos #96: Android phone camera to MotionEyeos

bomber23, 2016-01-01:
I recently posted the same problem on bitbucket, as an anonymous user , where I was instructed to upgrade to motioneyeos 0.27 and go here. I have done so but my problem persists. I am using the app 'IP Webcam Pro' to turn my old android phone (a samsung galaxy ace) into an ip camera. It can be accessed by typing in it's IPv4 address followed by the port in a web browser. However, when I try to add it to motioneyeos on my raspberry pi 2 I get the error: 'not a supported network camera'.

I was then asked what version of motion do I use, and if not sure I should post the output of the command motion -h, in which I get :motion Version Unofficial-Git-UNKNOWN, Copyright 2000-2005 Jeroen Vreeken/Folkert van Heusden/Kenneth Lavrsen

3. Thirdly I was told to post the url of my network camera (the samsung galaxy ace) which is: http://192.168.0.15:5100, that being it's IPv4 address. I can use that url in a web browser to access the camera.

Thanks in advance

ccrisan, 2016-01-01:
If you visit the URL you mentioned directly from a web browser you'll most likely be prompted with a nice UI. That is obviously not a raw MJPEG stream which is what motion(Eye) expects.

I installed myself the app you mentioned (or a similar one) and discovered that the MJPEG url is actually something like http://192.168.0.15:5100/video. Please try it like this and let me know if it works.

P.S. I hope you actually upgraded your motionEye to 0.30 which is the latest stable version, not 0.27 :)

bomber23, 2016-01-01:
Thank you for your help.
Using http://192.168.0.15:5100/video solved my problem and yes, I do have motionEye 0.30

closed 2016-01-01 by bomber23

barneyc, 2017-09-02:
Works a treat.  That's three old Android phones press-ganged into service :o)

wzwick, 2019-09-01:
2019 - info still helpful
thanks guys

mehdilaghrour, 2020-02-02:
hello

 im in rapsberry 3 
---------------------------------------------
motionEye Version | 0.41
Motion Version | 4.2.2
OS Version | motionEyeOS 20190911
-------------------------------------------
i would like use some old smartphone with android to be used like ipcam !
i have a same problem not same solution can you give me which apk your choice

keithmartinkinyua, 2020-03-24:
how can i add my phone camera to motion eye?

starbasessd, 2020-03-24:
https://play.google.com/store/apps/details?id=com.pas.webcam&hl=en_US
You would need to set the local IP address for the phone (easiest on router)
There are many such apps available.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

keithmartinkinyua, 2020-03-24:
Thank,,,,,is there an ios version for this?

starbasessd, 2020-03-24:
I don't know about that particular app, but yes, there are similar apps
available.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

keithmartinkinyua, 2020-03-25:
okay.thanks.
I hope you're staying indoors and sanitizing oftenly.
stay safe

RaghsMakes, 2021-03-28:
> okay.thanks.
> I hope you're staying indoors and sanitizing oftenly.
> stay safe

Droidcam