motioneye-project/motioneyeos #97: Fast netowrk cam slow

snagglewest, 2016-01-01:
After upgrading to 20151230 from 20151206 the fast network camera is very slow again.  Zoneminder shows 1.5 to 2,5fps.  Prior to that I was getting a solid 15fps with frame rate set at 15fps, 800x480 image and quality set to 25% in version 20151206 on a Raspberry Pi B+.
I initially did a upgrade but just to be sure I also completely reinstalled from a new image however the problem persists.  You might recall we had the same issue a few versions back in Motioneye too.

Snag

[logs.zip](https://github.com/ccrisan/motioneyeos/files/76009/logs.zip)

ccrisan, 2016-01-01:
Do you remember what exactly fixed that problem back then? Or at least point me to the discussion we had? It was motionPie, wasn't it?

snagglewest, 2016-01-01:
Yes, you're right, motionpie... If I recall correctly the issue "StreamEye fps drop to .5fps (#226)" but I cant seem to find the "issues" for motionpie any more.  If I recall you did a tweek and the next verion was working.  Sorry, not much help

Snag

ccrisan, 2016-01-03:
I have re-enabled issue tracking for motionPie (which I had disabled to prevent users creating issues for motionPie). This is the issue where we discussed this exact same problem: https://github.com/ccrisan/motionpie/issues/226. As far as I can see from that thread, I managed to fix the problem and haven't touched that code ever since.

Could you please try downgrading to the previous version (manually, or using another spare SD card) and see if the problem is indeed between the two versions?

ccrisan, 2016-02-21:
@snagglewest are you still experiencing the issue with streameye? What's the status of this problem? Should we still look into it?

bortek, 2017-02-23:
No answer from the use I will close this.

closed 2017-02-23 by bortek