motioneye-project/motioneyeos #98: Timestamp

dabi236, 2016-01-02:
I have motioneyeos successfully running on two Raspberry Pi 2's.  This first RPi board controls two cameras (RPi camera and a USB Logitech). The second RPi controls a waveshare night vision camera.  The system is a standalone and is not hooked to the internet.  When the RPi's boot up they are not able to get current date using http:.   Is there any way of setting the current date and time when not hooked to the internet?  I have attached a screen shot from the camera showing timestamp.

[image]

MarekITPro, 2016-01-02:
This is tracked on #71 
I get the same problem, it only works if you set it to use NTP instead of HTTP.

dabi236, 2016-01-02:
Thanks for the response.  I'm not sure where to change it to NTP.

dabi236, 2016-01-02:
Sorry found it

MarekITPro, 2016-01-03:
You can change it under Expert Settings/Date Method.
What version of motioneyeos are you running ?

dabi236, 2016-01-03:
I'm running version 20151206.  When I change it to NTP and apply the new setting the date is still 1969-12-31 and the time is also not correct.

MarekITPro, 2016-01-03:
Ehm, can you backup your config and update to 20151230 - my Raspi2 is running this latest build and with HTTP selected it updates the date about 10 minutes after booting up. 
For NTP setup I am assuming you get response when: "ping 0.pool.ntp.org".
Attach your messages.log and boot.log files - will take a look.

dabi236, 2016-01-03:
Upgraded to 20151230.  
With date method set to HTTP, I ran the camera for 15 minutes and there was no date/time correction. It shows 1970-01-01, 00:15:30.  When I set the date method to NTP I get the same results.  With the date method set to NTP I also typed in "ping 0.pool.ntp.org". I get the following response "bad address '0.pool.ntp.org"

[boot.docx](https://github.com/ccrisan/motioneyeos/files/76465/boot.docx)
[Motion.docx](https://github.com/ccrisan/motioneyeos/files/76466/Motion.docx)

ektus, 2016-01-03:
Without Internet, neither HTTP nor NTP have a chance to work as both need a connection. In this case, you would need a hardware clock with battery buffer and get the time from that one. The necessary command has been included in recent builds, but I don't know if it's been activated in the startup scripts yet as I've since switched to Raspbian Wheezy to try and use Pikrellcam in parallel.
https://github.com/ccrisan/motioneyeos/issues/61
https://github.com/ccrisan/motioneyeos/issues/45

Mikiya83, 2016-01-04:
Or you can use another computer in your private network as NTP server and usr it in your Pi, but it's a bit tricky.

Chiny91, 2016-01-04:
A bit tricky agreed but satisfying to implement it on another Raspberry Pi.
http://www.satsignal.eu/ntp/Raspberry-Pi-NTP.html

This setup turns out to be very reliable and quite capable of servicing a network of perhaps a dozen phones, tablets, TVs, gadgets, motionEyeOS... and a couple of external friends.

bortek, 2017-02-23:
Looks like this is no longer an issue. Please reopen otherwise.

closed 2017-02-23 by bortek

beankylla, 2019-08-27:
so i think the same underlying issue still applies: I want to setup a cron with a curl command to update a dydns entry, but the command does not work and i get a "bad address" message.
Trying to ping google.com or any web adress gives me a similar response.
pinging 8.8.8.8 works beautifully.
is anything preventind DNS resolve here?

Mistahbass, 2019-12-05:
I used NTP and this server :
ntpdate ntp.ubuntu.com

Cheers, Brock