motioneye-project/motioneyeos #99: TPLink wireless dongle not recognized

rjalexa, 2016-01-03:
Can anyone please help me troubleshoot this ?

I had to write the image using dd on a Max (getting EXT4 write with FUSE is a nightmare) and then added the ssid and psk lines to /etc/wpa_supplicant.conf after remounting the root filesystem in rw but I am under the impression that the drivers are not loaded at all

TPLINK TL-WN823N

thank you very much for any help and happy 2016.

ccrisan, 2016-01-03:
If you managed to boot your system *at least once* before editing `/etc/wpa_supplicant.conf`, then your change will not be taken into consideration. The reason is that this wpa_supplicant.conf file is used as a template and copied over to `/data/etc/wpa_supplicant.conf` (which is on an editable partition). See [this](https://github.com/ccrisan/motioneyeos/wiki/Tweaks#manually-editing-network-configuration) for more details.

The normal way of configuring your WiFi is to either boot with a network cable and temporarily use the wired connection, or to use the writeimage.sh script (if you hopefully use linux).

However, if you suspect your dongle is not compatible with motionEyeOS, please attach your log files here after booting with a wired connection (and recovering them from the UI) and then we'll see what's actually going on.

rjalexa, 2016-01-03:
No unfortunately I only have a Mac (other than the Raspberry), so I wrote the image to SD using dd and then connected an USB keyboard to the Raspberry Pi 2 to edit the files and configure the wifi network.

Reading the link you provided makes think that if I DELETE the /data/etc/wpa_supplicant.conf file upon the next boot it will be copied from /etc/wpa_supplicant.conf and used.

Is this correct ?

I am under the impression of having used that dongle uses rtl8192cu as the driver and that I had set it up on a prior installation of MotionEyeOS.

What commands can I use to see if it has been recognized ? Which logs would you need to look at to understand ?

Thank you very much.

ccrisan, 2016-01-03:
Yes, you're right. If you manage to delete the `/data/etc/wpa_supplicant.conf` file, the system one will be used at the very next boot. If you're positive the driver used is `8192cu`, then there should be no problem with your device, it should work without any additional tweaks.

If you manage to type in some commands using the keyboard, here are a few commands that could be of use:

 * list loaded modules (i.e. drivers) - look for 8192cu

         lsmod

* list connected usb devices

        lsusb

* look through the kernel log for your device:

        dmesg | grep 8192cu

* look through the boot log to see what the wifi network script has to say (use arrows to scroll, q to quit):

        less /var/log/boot.log

Chiny91, 2016-01-03:
**rjalexa** *unfortunately I only have a Mac* ?    Sir, you have a most fine computer !   Should you in future need, or even wish, to re-write that motionEyeOS image, you might find it easier to use Apple PiBaker http://www.tweaking4all.com/hardware/raspberry-pi/macosx-apple-pi-baker/ .   That app also makes it simple to backup the entire system, including any config files.   Certainly I found it useful when my SDcard recently died.

rjalexa, 2016-01-04:
Sirs thank you very much.
Copied my /etc files onto the /data/etc ones and rebooted and voilà all is good.

The "unfortunately" bit refers only to not being able to use the VERY useful script (the FUSE/EXT2 stuff gave me so much pain in the past) hence having to do some minimal manual editing, which once understood is really simple.

I had found the very cool pi-baker app which I had actually used instead of the manual dd to dump the image onto the SD on the Mac. Very useful and believe it or not did not think about the even MORE useful function to backup my SD which I'll surely do.

I'll now try to put motioneyeos to good use, after I find a way to protect the PI and its camera from rain :-)

Happy 2016

closed 2016-01-04 by rjalexa

causeghs, 2017-01-25:
Hi all,
@rjalexy : I have the exact same problem with the TP-Link WN-823N. Motioneyeos says: wpa_supplicant: no device
When connected with a wire i have perfect connection. Please tell me how you fixed this Problem. Unfortunately i dont have mac/linux to usenthe writeimage script.

rjalexa, 2017-01-25:
Unfortunately I have stopped trying with that and have bought an Ethernet over electricity adapter.
Sorry for not helping more.
Bob

ccrisan, 2017-01-25:
@causeghs does your wifi dongle work with Raspberry PI + Raspbian?

causeghs, 2017-01-25:
@ccrisan yes it does

ccrisan, 2017-01-25:
@causeghs then it will work with motionEyeOS as well. Either a recent kernel update added support for it or I left out some firmware required for this dongle. Could you please identify it using `lsusb`?

causeghs, 2017-01-26:
@ccrisan Ok nevermind, it does not work on raspbian, thought it would since it said linux support..
I see that many people have problems with this dongle, also trying to install the RTL8192CU chipset driver is not working, i keep getting header errors..

I guess the best solution would be to send it back since i just got it yesterday via amazon. Can you suggest a dongle which works fine?

orlandperz, 2017-09-20:
Hi, ccrisan i have the same problem TPLink wireless dongle not recognized, 
the following are the results of the commands described above, 
I hope you can help me.

```
[root@meye-9553e34a ~]# uname -a
Linux meye-9553e34a 4.9.41 #1 Sun Aug 27 02:14:43 EEST 2017 armv6l GNU/Linux
```

```
[root@meye-9553e34a ~]# lsmod
Module                  Size  Used by    Not tainted
ipv6                  384500  0 [permanent]
evdev                  11746  0
spidev                  7034  0
rtl8xxxu               90576  0
mac80211              650739  1 rtl8xxxu
cfg80211              525838  1 mac80211
rfkill                 21373  1 cfg80211
snd_soc_bcm2835_i2s     7257  0
regmap_mmio             3977  1 snd_soc_bcm2835_i2s
snd_usb_audio         143956  0
snd_hwdep               6479  1 snd_usb_audio
snd_usbmidi_lib        22479  1 snd_usb_audio
snd_soc_core          175776  1 snd_soc_bcm2835_i2s
snd_rawmidi            23727  1 snd_usbmidi_lib
snd_seq_device          5266  1 snd_rawmidi
snd_compress           10144  1 snd_soc_core
snd_pcm_dmaengine       6002  1 snd_soc_core
snd_bcm2835            23131  0
uvcvideo               80130  1
snd_pcm                97825  5 snd_soc_bcm2835_i2s,snd_usb_audio,snd_soc_core,snd_pcm_dmaengine,snd_bcm2835
snd_timer              22702  1 snd_pcm
snd                    68784 10 snd_usb_audio,snd_hwdep,snd_usbmidi_lib,snd_soc_core,snd_rawmidi,snd_seq_device,snd_compress,snd_bcm2835,snd_pcm,snd_timer
i2c_bcm2835             6433  0
spi_bcm2835             7424  0
bcm2835_gpiomem         3791  0
uio_pdrv_genirq         3718  0
uio                    10166  1 uio_pdrv_genirq
fixed                   3029  0
i2c_dev                 6642  0
i2c_bcm2708             5740  0
bcm2835_v4l2           43131  0
v4l2_common             6285  1 bcm2835_v4l2
videobuf2_vmalloc       5849  2 uvcvideo,bcm2835_v4l2
videobuf2_memops        1939  1 videobuf2_vmalloc
videobuf2_v4l2         16841  2 uvcvideo,bcm2835_v4l2
videobuf2_core         31781  3 uvcvideo,bcm2835_v4l2,videobuf2_v4l2
videodev              177734  6 uvcvideo,bcm2835_v4l2,v4l2_common,videobuf2_v4l2,videobuf2_core
media                  28994  2 uvcvideo,videodev

```
```
[root@meye-9553e34a ~]# lsusb
Bus 001 Device 001: ID 1d6b:0002
Bus 001 Device 005: ID 0458:706e
Bus 001 Device 003: ID 0424:ec00
Bus 001 Device 002: ID 0424:9512
Bus 001 Device 004: ID 0bda:818b

```
```
[root@meye-9553e34a ~]# dmesg | grep 8192
[    0.000000] PID hash table entries: 2048 (order: 1, 8192 bytes)
[    0.000000] Memory: 370956K/393216K available (5966K kernel code, 490K rwdata, 1956K rodata, 396K init, 733K bss, 14068K reserved, 8192K cma-reserved)
               [    5.963212] usb 1-1.2: rtl8192eu_parse_efuse: dumping efuse (0x200 bytes):
[    5.963920] usb 1-1.2: RTL8192EU rev B (SMIC) 2T2R, TX queues 3, WiFi=1, BT=0, GPS=0, HI PA=0
[    5.963933] usb 1-1.2: RTL8192EU MAC: 00:e0:4c:10:ea:bf
[    5.963946] usb 1-1.2: rtl8xxxu: Loading firmware rtlwifi/rtl8192eu_nic.bin
[    5.967157] usb 1-1.2: Direct firmware load for rtlwifi/rtl8192eu_nic.bin failed with error -2
[    5.967188] usb 1-1.2: request_firmware(rtlwifi/rtl8192eu_nic.bin) failed

```

```
[root@meye-9553e34a ~]# less /var/log/boot.log
---- booting motionEyeOS 20170827 ----
 * Detecting disk device: /dev/mmcblk0
 * Mounting filesystems: done
 * Loading kernel modules: done
 * Setting hostname: done
 * Starting syslogd: done
 * Configuring CPU frequency: done
 * Starting eudev: done
 * Starting watchdog: done
 * Starting wpa_supplicant: no device
 * Configuring wireless network: no device
 * Configuring wired network: dhcp
 * Setting current date using http: Wed Sep 20 08:11:10 COT 2017
 * Starting http date updater: done
 * Starting crond: done
 * Starting sshd: done
 * Starting proftpd: done
 * Starting smbd: done
 * Starting nmbd: done
 * Starting motioneye: done
 # Interface eth0 has IP address 192.168.0.23/24
 # Default gateway is 192.168.0.1
 # DNS server address is 190.157.8.33

```

I tried the solution that appears in several forums but the commands sudo and apt-get are not recognized, I have also tried to install the drivers that come in the dongle cd but I get errors. Thank you for the help you can give me.

ccrisan, 2017-09-20:
@orlandperz could you please try copying the following file to `/lib/firmware/rtlwifi/` on your motionEyeOS system:
[rtl8192eu_nic.bin.zip](https://github.com/ccrisan/motioneyeos/files/1319132/rtl8192eu_nic.bin.zip)?

If it works, I'll make sure to include the file with the next release.

P.S. You'll need to `mount -o remount,rw /` before copying it.

orlandperz, 2017-09-20:
Thanks for your quick response. I'll put my hands on the job and let you know the results.

orlandperz, 2017-09-20:
Calin, I followed his instructions and it works, however the pi now has two IP addresses as you can see:

```
---- booting motionEyeOS 20170827 ----
 * Detecting disk device: /dev/mmcblk0
 * Mounting filesystems: done
 * Loading kernel modules: done
 * Setting hostname: done
 * Starting syslogd: done
 * Configuring CPU frequency: done
 * Starting eudev: done
 * Starting watchdog: done
 * Starting wpa_supplicant: done (rtl8xxxu)
 * Configuring wireless network: dhcp
 * Setting current date using http: Wed Sep 20 16:58:27 COT 2017
 * Starting http date updater: done
 * Starting crond: done
 * Starting sshd: done
 * Starting proftpd: done
 * Starting smbd: done
 * Starting nmbd: done
 * Starting motioneye: done
 # Interface eth0 has IP address 192.168.0.23/24
 # Interface wlan0 has IP address 192.168.0.20/24
 # Default gateway is 192.168.0.1
 # DNS server address is 190.157.8.33

```

i Disconnect the ethernet cable and when accessing ip access the wlan0 not allow me, you have some idea about what is happening, thanks again.

ccrisan, 2017-09-21:
@orlandperz obviously it has two IP addresses as there are two network interfaces available and connected. After plugging out the ethernet cable, have you tried accessing it via the wlan0 IP address? Also, can you attach a monitor and see how the boot log looks like without a wired connection?

orlandperz, 2017-09-21:
Calin, turn off the pi, disconnect the ethernet cable and connect it again, this is the log:

```
---- booting motionEyeOS 20170827 ----
 * Detecting disk device: /dev/mmcblk0
 * Mounting filesystems: done
 * Loading kernel modules: done
 * Setting hostname: done
 * Starting syslogd: done
 * Configuring CPU frequency: done
 * Starting eudev: done
 * Starting watchdog: done
 * Starting wpa_supplicant: done (rtl8xxxu)
 * Configuring wireless network: dhcp
 * Setting current date using http: Thu Sep 21 08:58:16 COT 2017
 * Starting http date updater: done
 * Starting crond: done
 * Starting sshd: done
 * Starting proftpd: done
 * Starting smbd: done
 * Starting nmbd: done
 * Starting motioneye: done
 # Interface wlan0 has IP address 192.168.0.20/24
 # Default gateway is 192.168.0.1
 # DNS server address is 190.157.8.33

```

and it works very well, thank you very much for your help, support and quick response, I will follow your project. I hope to learn more about pi and embedded systems, really can do many very interesting things. Sorry for my English, it's not my native language. Thanks again and I hope the new versions of MotioneyeOS great work.

ccrisan, 2017-09-21:
@orlandperz few people around here are native English speakers. As long as your text is intelligible, we can help you :)