motioneye-project/motioneyeos #100: Wifi (8192cu) connection drops after a few minutes

rjalexa, 2016-01-04:
Dear friends,
now that I am connected by wireless I see the wifi connection dropping after a few minutes with messages about wlan0 having no IP address and then after 20 seconds motioneye reboots
and the cycle repeats.

Off the top of my head I seem to remember that the 8192 module has an issue with power management and in some other system I think (sorry my memory is not good) I had solved it by creating

/etc/modprobe.d/8192cu.conf

and inserting the following lines it it:

# Disable power management
options 8192cu rtw_power_mgnt=0 rtw_enusbss=0

as per this post:

https://www.raspberrypi.org/forums/viewtopic.php?t=61665

I am still not sure if I should try this in the /etc directory or in the /data/etc

Help please.

ccrisan, 2016-01-04:
The power management "fix" has already been included in motionEyeOS for a long time. Please attach your log files here so I can see what's going on.

rjalexa, 2016-01-05:
I am under the impression this MIGHT be an hardware related problem (power?) since I don't see anything strange in the logs, but I'll let your superior knowledge take a look too.

From an HW point of view it's a RPi2 with the Raspberry power adapter and camera module without IR filter with a USB attached SSD 120GB and a USB TP-Link dongle.

Thank you.

Here are the logs:

[motion.log.txt](https://github.com/ccrisan/motioneyeos/files/78800/motion.log.txt)
[motioneye.log.txt](https://github.com/ccrisan/motioneyeos/files/78801/motioneye.log.txt)
[boot.log.txt](https://github.com/ccrisan/motioneyeos/files/78803/boot.log.txt)
[dmesg.log.txt](https://github.com/ccrisan/motioneyeos/files/78804/dmesg.log.txt)
[boot.log.txt](https://github.com/ccrisan/motioneyeos/files/78806/boot.log.txt)

ccrisan, 2016-01-05:
Is there any particular reason why you haven't posted your messages.log file? It actually contains a very useful piece of information: brownout detections.

ccrisan, 2016-01-05:
Also your dmesg.log file contains many USB-related connection/disconnection events which usually indicate problems with your power supply. Please try removing your USB-attached SSD drive and see if the problem persists. Other idea would be to use a more powerful PSU.

rjalexa, 2016-01-05:
Sorry I must have overlooked. 

I am totally clueless about electronics :-( I tried another PSU from Ebay which was rated 3A but I am under the impression vendors over-rate them to get higher prices :-( Can you suggest anything good which I could buy off Amazon or Ebay in Italy or Europe ? Thanks again

[messages.log.txt](https://github.com/ccrisan/motioneyeos/files/78873/messages.log.txt)

ccrisan, 2016-01-05:
Well 3 Amps should be more than enough. Try to see how it behaves w/o the SSD first.

P.S. your messages file doesn't say anything about brownout so it may be that the USB controller is somehow overwhelmed. Anyway, again, try removing the SSD.

rjalexa, 2016-01-05:
Ok thanks. In case the problem is the SSD what is the suggestion to store quite a large number of videos ? Would a 32 - 64 gb usb stick draw less power ?

ccrisan, 2016-01-05:
In that case, using a powered USB hub might be the best solution.

ektus, 2016-01-05:
Normal USB sticks or large SD cards (64GB MicroSD are available, after all) will work, but are not designed for heavy use and have been reported to fail after some time (often less than a year) when used in a surveillance application that writes lots of data. Also, especially with videos, they may severely limit your storage time before deleting old files becomes necessary.

I'm going a different way. The RPi 2 has enough USB ports, but limited power to these ports. So I've gone the "dirty" way to supply power: Soldering a "Y" cable with a couple connectors that connect the power supply to **both** the micro USB power inlet of the RPi **and** one of the standard USB ports. That way, there's enough power at the RPi's internal USB hub to drive a 1TB Toshiba 2.5" (USB 3 without separate power connector) hard drive that's going to provide ample storage room for pictures and videos. I don't need a separate, powered hub for this. Don't know if it will work long-term, but testing for the last couple hours looks good, and voltage is stable (red power LED constantly lit, no rainbow square on the attached monitor). I had to solder, though, as the adapter cable I've got off Amazon (Delock brand) was too weak. Even with a good power supply, many USB cables are just too flimsy to transport the power required without a noticeable voltage drop.

rjalexa, 2016-01-06:
VERY interesting. Thank you.
BTW now that you mention I remember having seen a multicolored square on the screen and could not understand where it came from :-) Meaning ?
Powered USB would be connected to both one of the RPi ports and the SSD Disk right and/or WiFi key right ?

UNRELATED questions:
a) Is there a way to suppress reboot if there's no network ? If I only have local storage and no connection for example.
b) Is there a forum to ask for help / opinions or this is the right place even not for problems but just suggestions ?
c) Is MotionEyeOS based on some known Linux distro or is it custom ?

Thanks

ccrisan, 2016-01-06:
The rainbow rectangle shows up when the voltage drops below a certain limit (4.75V I think).

a) Go to Expert Settings and disable the Network Link Watch option.
b) There always has been [this thread](https://www.raspberrypi.org/forums/viewtopic.php?f=43&t=83369), but normally you should open issues for problems as well as for suggestions.
c) Both. motionEyeOS is a custom distro based on [buildroot](https://buildroot.org/).

I will close the issue now as it is clearly related to the power.

closed 2016-01-06 by ccrisan

ektus, 2016-01-06:
Power goes to the Micro connector for the board and one of the four USB connectors as additional supply for the USB hub. Your USB SSD and your USB WiFi stick will plug into two of the remaining USB connectors. Take care to use short cables for USB power, the shorter the better. Any PSU, even a 3A one, may have an output voltage anywhere in the range of 5V +-5%, that's a range from 4,75V to 5,25V and the lower end being the lowest acceptable value for normal operation of the RPi. Bad cables will cause an additional voltage drop, so the voltage available at the USB port may fall way below acceptable levels. The RPi itself doesn't care (much), but your peripherals may malfunction because of the low voltage condition. The easiest way is to add a powered USB hub and connect your USB devices to that one, but this needs a second PSU (or just a PSU with two USB outlets, one cable to the RPi and one to the hub if you can find the right cable).