motioneye-project/motioneyeos #103: Can't change hostname

phibes4000, 2016-01-05:
I have tried setting at /etc/hostname, /data/etc/hostname, via the hostname command, and the hosts file (per board/common/overlay/etc/init.d/S01hostname) but cannot get the change to survive a reboot.  It defaults back to "$os_prefix-$board_sn".  Am I missing something?

ccrisan, 2016-01-05:
Hostname is not yes changeable. It will be soon, though.

Arut42, 2016-01-07:
Thats easy, login with ssh and write the folowing:

echo "hostname" > /data/etc/hostname
mount -o remount -w /dev/root /
mv /etc/init.d/S01hostname /etc/init.d/S05hostname
mv /etc/init.d/S13mountall /etc/init.d/S01mountall 
sync
reboot

The problem is, set hostname before mount
/etc/init.d/S01hostname

sys_hostname_file="/etc/hostname" -> not mount, can't find file
hostname_file="/data/etc/hostname" -> not mount, can't find file

test -n "$os_version" || source /etc/init.d/base -> default maye-serial

case "$1" in
    start)
        msg_begin "Setting hostname"

        if [ -f $hostname_file ]; then -> file not found
            hostname=$(cat $hostname_file)
        elif [ -f $sys_hostname_file ]; then -> file not found
            hostname=$(cat $sys_hostname_file)
        else
            hostname="$os_prefix-$board_sn" -> default
        fi
        /bin/hostname $hostname
        echo "127.0.0.1 localhost $hostname" > /etc/hosts

birukun, 2016-01-09:
Arut42 I tried that on my implementation but the root file system is locked so it did not stay persistent.  I believe it will be possible to do if you mount the filesystem elsewhere read/write and make the modification.  I have not tried that yet, playing with the dozens of other fun features first :-)

Arut42, 2016-01-09:
this command remount writeable:

mount -o remount -w /dev/root /

don't forget space between /dev/root   and  /

best regards

ccrisan, 2016-01-09:
This procedure won't be needed anymore starting with 20160109; you will only have to create the file `/data/etc/hostname` containing your desired hostname and reboot.

closed 2016-01-09 by ccrisan

ccrisan, 2016-02-21:
I've just committed the changes required to allow setting a custom hostname from the UI. The feature will be available in the next release.

joshhighet, 2016-09-16:
Hey @ccrisan , Thanks for all the work you have done on Motioneye, a great accomplishment. How exactly do I change the hostname over the UI?

ccrisan, 2016-09-16:
@joshhighet Settings Menu -> General Settings > Hostname.

ugoi, 2023-10-04:
What is the default hostname?

This doesnt work:

ssh pi@raspberrypi.local 

ping raspberrypi.local

starbasessd, 2023-10-04:
You could look on your router/modem to see what IP address was issued to the Pi (RPis mac addresses begin):
[image]