motioneye-project/motioneyeos #104: How to change the network SSID

fredericcollin, 2016-01-05:
Hi, I have several Motioneye pi's on my network and have a hard time distinguishing them with their default names such as MEYE-A8AE5E340, etc...How can I rename them to something more intuitive such as 'entrance', 'garage', etc...?
Thanks in advance!

phibes4000, 2016-01-05:
Nothing to do with SSID.  It's the hostname.  I asked the same question 12m prior - https://github.com/ccrisan/motioneyeos/issues/103.

fredericcollin, 2016-01-05:
Hi phibes4000, of course you are correct! I do indeed mean 'hostname', thanks.

ccrisan, 2016-01-05:
Hostname is not yes changeable. It will be soon, though.

fredericcollin, 2016-01-07:
Thank you for your kind reply Crisan, looking forward to it!

closed 2016-01-07 by fredericcollin

ccrisan, 2016-02-21:
I've just committed the changes required to allow setting a custom hostname from the UI. The feature will be available in the next release.

fredericcollin, 2016-02-22:
Excellent, thanks!

https://github.com/ccrisan/motioneyeos/issues/104#issuecomment-186808798

ioquatix, 2017-09-10:
How do you do this?

ccrisan, 2017-09-11:
@ioquatix in Expert Settings you have a Hostname setting.

alexxtasi, 2019-06-21:
Based on https://github.com/ccrisan/motioneyeos/issues/103#issuecomment-247540487 and my little experience, I  think it's:
_Settings Menu -> General Settings > Hostname_

ioquatix, 2019-06-21:
Oh, that's awesome!