motioneye-project/motioneyeos #105: Feature request: Video with audio

MrTinkerman, 2016-01-05:
Hi,

I tried MotionEyeOS on a Raspberry Pi2 and it is really all one would want in a surveillance system !
Very nice, easy setup, clean, all the right options, rtsp and webhook support, awesome :)

One thing thouh, the recorded video is without audio :(
I looked into Zoneminder and it also has no audio recording. Is this difficult to implement ?

Anyway, I hope MotionEyeOS could support video with audio at some point.

Thanks for all the great work so far :)

MrTinkerman, 2016-01-05:
I see the audio support is in the roadmap, among other cool features.

I hope you find the time to implement all this.

Thanks

ccrisan, 2016-01-06:
Audio support will at least be coming in its "live" form (i.e. no recording support). Live audio could be implementing without touching motion's source code and will take advantage of the HTML5 web apis.

zettam, 2016-04-01:
Any updates with this? When should we expect this?

ccrisan, 2016-04-01:
Not any time soon as I'm caught with other stuff these days. Sorry.

aplocher, 2016-06-15:
Hello, don't meant to be a pest but I was wondering if there was any updates on this?  Quite frankly, I couldn't care less about audio during recording - just live streams mainly.  Is this possible?

I'm prototyping a baby monitor, and so far after trying Raspbian with VLC, FFMPEG, etc - this is my favorite ... except for that one little (or big) feature that I need.  

Thank you!

danielhoherd, 2016-12-18:
Having audio play over the stream would be awesome. This could be a fantastic baby cam if that feature was added.

ccrisan, 2016-12-18:
There's no audio support at all and it is very unlikely this could ever happen, given that it has to be implemented in [motion](https://github.com/Motion-Project/motion) first.

czarina, 2017-01-08:
I've been searching everywhere for a system like motioneyeos, but that supports audio streaming - I'm at my wit's end! How difficult would it be for me to try to add audio support to motioneyeos myself? Does anyone know of other libraries like motion but that include audio support?

ccrisan, 2017-01-08:
It might be the project of your life, but you could try implementing it. You'll need to:
1. implement audio capturing support in motion
2. rewrite motion to support h264 streaming instead of (or along with) mjpg
3. rewrite motionEye so that it uses h264 instead of mjpg internally
4. rewrite motionEye's UI streaming part so that it uses h264
5. test everything and create pull requests

ryny24, 2017-01-25:
This os is amazing, but audio is a requirement. I'll keep an eye on this thread..

buzink, 2017-02-05:
Could you not just do something like `ffmpeg -f alsa -ac 1 -i hw:1 -ar 44100 -f mpegts udp://10.255.11.53:1234` ?

buzink, 2017-02-12:
@ccrisan Is it not as simple as 
1. compiling ffmpeg with alsa and mp3 support
2. starting command `ffmpeg -f alsa -ac 1 -re -i hw:0,0 -acodec libmp3lame -ab 32k -f rtp rtp://192.168.178.51:1234`
3. embedding `rtp://192.168.178.51:1234` in a html5 audioplayer in the html page?

This will not create a combined video/audio stream, but it will provide both video and audio streams and the ability to play them at the same time.

ccrisan, 2017-02-13:
That would capture audio in (almost) real time, which could be useful for some, but most will complain about not having it embedded in the recorded movies.

jasaw, 2017-08-23:
I think audio can be added in live stream or recorded movie by using motion extpipe to ffmpeg. From ffmpeg command, you can easily stream the rtsp output to anywhere you like.

jasaw, 2017-08-23:
I haven't got time to try this out yet, but here's what I would do based on my experience from my own [baby monitor](https://github.com/jasaw/bbPiCam) project from 3 years ago.
1. Enable alsa audio driver and audio tools (alsamixer, arecord, ...) in MotionEyeOS.
2. Patch MotionEye to allow extpipe. It's a 2 liner patch.
3. Enable AAC encoder in MotionEyeOS. Last time I used faac-1.28.
4. Patch ffmpeg to reduce max-interleave-delta, otherwise audio and video eventually drifts too far apart and loses sync. I had to patch it because the command line option didn't work, not sure if it has been fixed by now.
5. Add to ffmpeg configure options: --enable-libfaac
6. Once compiled everything, it's time to play with motion's extpipe. Something like this extpipe should encode video and audio into audio video interleaved stream: 
`ffmpeg -fflags +nobuffer -re -y -f rawvideo -pix_fmt yuv420p -video_size %wx%h -framerate %fps -i pipe:0 -fflags +nobuffer -re -f alsa -ar 16000 -ac 2 -i hw:1,0 -map 0:0 -map 1:0 -c:v h264_omx -profile:v high -b:v 2000000 -c:a aac -b:a 16k -ac 1 -f rtsp -metadata title=BabyMotionEye rtsp://0.0.0.0:554`. To record into movie instead of RTSP stream, replace `-f rtsp -metadata title=BabyMotionEye rtsp://0.0.0.0:554` with `-f mp4 %f.mp4` or `-f matroska %f.mkv`.
7. Disable motion's pre-capture and motion gap as they will cause havoc to the video stream.
8. For streaming, I used RTSP and RTMP protocol for low latency and best overall playback support from web browsers and mobile devices. 
9. Install crtmpserver on MotionEyeOS to convert RTSP to RTMP if you want RTMP.
10. Patch MotionEye to serve the video audio interleaved stream.

LTechik, 2017-09-05:
@jasaw have u tried it yet ?
All my tries to get video and sound simutaniously on logitech usb camera with integrated mic ended bad, best option for that was vlc or u4vl, but both options lack stuff motioneye do.

jasaw, 2017-09-05:
@LTechik Sorry no, I haven't got any time to try this. I already have my home made baby camera from 3 years ago, so this is really low priority for me.

I wonder whether MotionEye(OS) really the right place to start if you want a baby monitor. Motion(Eye(OS)) really is geared towards surveillance, which is quite different from a baby monitor. I did a thorough comparison 3 years ago, and concluded that motion software is not the best choice for a baby monitor. I ended up with raspivid + ffmpeg + nginx + crtmpserver. What it lacked was a nice web interface.

LTechik, 2017-09-06:
I got you, does ur baby monitor works with logitech cams with build in mic ?
Thanks for your answer !

jasaw, 2017-09-06:
@LTechik I've only used the Pi camera. If you want to use a different camera, try changing the ffmpeg command to use the USB camera as video input. More info on how to use ffmpeg with USB camera [here](https://trac.ffmpeg.org/wiki/Capture/Webcam). Same thing for audio.

jordanwilliams1, 2019-07-02:
3 Years since this has feature quest and it's still not added :( Damm I really wanted audio for my camera.

jasaw, 2019-07-02:
There's audio support in Fast Network Camera mode with RTSP stream. See https://github.com/ccrisan/motioneyeos/pull/1764
Keep in mind that the audio part is not well tested. I suspect there's a problem with audio and video drift, but I haven't got time to fully test it.

To record videos with audio, you'll have to rely on motion's extpipe option and use ffmpeg to add in the audio channel, as I have mentioned in my earlier post. Alternatively, you could create a feature request in motion project and hope for the best. 3rd alternative is to implement audio encoding support in motion and do a pull-request to get it merged upstream.

rapattack55, 2020-01-19:
Definately would like to see audio in motioneyesos....didnt realise motion didnt have it bummer. I really need it so i can catch who is making all the noise at my apartment block not just see them

jasaw, 2020-01-20:
@rapattack55 [Motion](https://github.com/Motion-Project/motion) developers are reluctant to support audio recording because of privacy laws in certain countries. Maybe motion developers will change their mind when there's enough people requesting for audio feature?

rapattack55, 2020-01-28:
Not sure i responded to this....anyway yes seems so. I would like to request it. There are sections of the law it can be used. But depends where you are as well. I am in australia. So theres many police forces or agents that wont look at security footage at all but some will in some cases. I dont think it can be used in evidence in court but the other day i caught someone in the backyard common area stealing some ornaments in the garden. I reported this to a section call 'police assistance'. A phone line thing. Anyway they were straight away saying they want to see the footage. In other cases to do with housing(government) it always depends on the staff member as to wether they will look at an anti social neighbour doing something wrong on footage i ask them to view. One particular manager was on it and viewed in his office while i was there. It didnt go to eviction of that tenant at that time but it put doubt in their minds that this tenant was all he said he was. Then theres 000 emergency police. They almost never view footage from a security camera. In all cases none care about audio. It never comes up.

zhuravel-sergey, 2020-02-12:
3 years have passed, and they did not add audio recording :(

starbasessd, 2020-02-12:
More than likely, audio streams vary too much from camera to camera, and
require a lot more programming than what was worth CCrisan's time and
effort. I'm quite happy that this project is working as well as it is, and
across such a wide spectrum of cameras, and SBC hardware, personally. I
wish I could afford to buy more SBCs and cameras, so I could better help
support end-users. I hope that the people I've supported let CCrisan know
that his project is well worth it.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

grtessman, 2020-04-11:
I just installed motioneye... My search for why sound/microphone support wasn't working led me here.  Would be a great feature to add!

DeadEnded, 2020-05-29:
I know audio in MotionEye can be a hot topic.  I have been using MotionEye for 24/7 recording and it has been fantastic!  As I wanted my recordings to have audio, I was able to write up a script to do it.

I am a novice at best, and it probably is not the cleanest or best way to do it... but it works.  It basically kicks off an audio recording whenever a video recording starts (yes an extra process).  Then when video stops, it stops audio recording and then merges the two together.

I have not testing it beyond my single camera model doing 24/7 recording in 60 minute segments... but it should work for others as well.  I don't want to break any rules or anything, so I will first ask if there is any issue with me posting a link here to it?

If there is not a problem with posting a link to it I'm happy to share it... again its a novice's work - but help to improve it will always be welcome (PR's)!

Cheers!
DeadEnd

MacDos, 2020-05-31:
Has someone tried the Modul "ReSpeaker 2-Mic Pi HAT V1.0 for Raspberry Pi Zero and Zero W"
Sorry, I have no software skills, but perhaps someone could use it?

Tylersuard, 2021-01-26:
@DeadEnded I'm interested!  Can you share the script please?

DeadEnded, 2021-01-26:
https://github.com/DeadEnded/MotionEyeAudio

I am no longer using MotionEye, but can merge any fixes that others find... or you can fork it.
Good Luck!

rapattack55, 2021-01-26:
Looks good. Can anyone do a beginner instruction on this? I dont understand much i am afraid. Carla

 
 

https://github.com/DeadEnded/MotionEyeAudio

JW6677, 2021-02-10:
I hope this gets implemented into the core of motion eye os and works out of the box.

DeadEnded, 2021-02-10:
Unlikely.
MotionEye is based on the motion daemon.
It has been stated multiple times that until audio is added to motion, MotionEye won't have it.
What I created was a hack at best... it has many flaws and is a workaround.
I have since moved on to another NVR software that does support audio out of the box.

0BackDoor1, 2021-05-16:
Discovered Meye about a week ago, installed and it's very amazing project, works very well, until "hey, wait! where is audio settings?" and then the bad news, even if rtsp has audio info, meye simply doesn't use it...  what's the meaning of a NVR system without audio???

starbasessd, 2021-05-16:
@0BackDoor1 
N Network
V VIDEO
R Recorder
<sigh>

0BackDoor1, 2021-05-16:
> @0BackDoor1
> N Network
> V VIDEO
> R Recorder

oh thanks man! I never read meaning of "video" word before, I'm going to write to youtube's support because on their VIDEO they put also AUDIO, that's crazy thing!

starbasessd, 2021-05-16:
@0BackDoor1 
It's still open because it's a feature request.
It's been requested multiple times in multiple threads.
I was accepted as a support person last year, and have been slowly going through the open tickets, closing ones completed, trying to close those that haven't had any follow ups, and labeling the ones kept open to make it easier to find active issues, enhancements and feature requests for anyone who wants to contribute. 
The owner of the project hasn't put much effort into it maybe because he doesn't see any priority for it. There needs to be some upstream support in motion (which he contributes, but is not the owner) before motionEye will support it. Look at his first responses to this thread.
Are you also going through Motion-Project's support issues and 'bumping' them to get audio support?
He has made it abundantly clear that if you want to implement it, feel free. Motion and motionEye are both open source, and will accept patches, fixes, new features.