motioneye-project/motioneyeos #107: Additional information in the wiki: services menù

lozziboy, 2016-01-06:
I found that FTP, SAMBA, SSH servers can be enabled under the services menù, but for me is not clear what they are and what use i can do.
I ask for a section in the wiki, with a short explanation (and example scenario) about them.

[image]

Calcyfer, 2016-01-06:
A quick and very condensed explanation in the meantime : 
* FTP server allows access to the filesystem over FTP (file transfer protocol) on port 21
* Samba server allows access from windows and linux machines to the filesystem (Samba is a network protocol)
* SSH server allows console access (using f.e. Putty) on port 22 so you can configure your system remotely without using the web front end

Chiny91, 2016-01-08:
Samba also works nicely with Macs :apple: and I'd be surprised if not with other OSs as well.  

I use only ssh as I find it does all that I require.   Maybe not running ftp & samba servers  saves a small amount of cpu oomph.

Shame rsync is missing <<-- Hint

closed 2017-02-23 by bortek