motioneye-project/motioneyeos #109: Please advise best settings for Pi2B and Pi Camera

rjalexa, 2016-01-06:
My setup is a Raspberry Pi 2 Model B with a Raspberry Pi Camera Module without IR filter.

This setup is meant to capture and store 5-10 max 5 minutes videos and store them on a 120GB SSD for max 1 month.

Can anyone please advise the best settings amongst or explain what process to find them myself.

Video resolution (currently 1600x1200 which is the max I see in the interface even though I would have liked FullHD. The camera supports 2592 x 1944 pixels AFAIK) ?

Framerate ?

Movie quality ?

GPU Memory ?

Overclocking (currently experimenting with Pi2 setting). Is it known to be stable ?

Thank you very much

Crazyslade, 2016-01-07:
I have the same setup as you and would also like to know the settings.  I seem to only be getting 2-3 fps in the preview window and streaming to a browser window. I have the settings set to 30 fps but no change.  Is this the same for you rjalexa?

rjalexa, 2016-01-07:
Same here my friend. Let's see if someone with more experience chimes in. Happy 2016.

ccrisan, 2016-01-07:
You won't get a better fps with resolutions as high as 1600x1200.

rjalexa, 2016-01-07:
Just to understand I have lowered the resolution of the camera to only 1024x576.
What is the meaning of the two frame per second values overlayed on the bottom left of the video ?
For example I now have 25.0/15.0 fps with the frame slider set to 15.

It seems that when motion is detected one of those two values drops significantly even to only 2-3 fps !!! Why ?

Crazyslade, 2016-01-08:
@rjalexa Happy 2016!
@ccrisan  I have used the lowest resolution selectable and still the fps stays the same.

ccrisan, 2016-01-08:
@Crazyslade the camera and the device work perfectly together at high resolutions and framerates, but not when the entire stream passes by and is processed by the motion daemon. Any additional processing on the frames will eat up a considerable amount of CPU. If you want to get the maximum performance from your PI + CSI Camera, enable the [Fast Network Camera](https://github.com/ccrisan/motioneyeos/wiki/Fast-Network-Camera) option.

rjalexa, 2016-01-08:
@ccrisan Would it be possible to use the Fast Network Camera option paired with a PIR sensor such as http://www.amazon.it/gp/product/B0104LSJGQ?psc=1&redirect=true&ref_=oh_aui_detailpage_o04_s00 to have both a fast capture and store only when motion (by the PIR) is detected ?

Mikiya83, 2016-01-08:
I just buy PIR module, with a script at startup, i'm pretty sure we can enable/disable motionEye according to PIR detection (in a python script that monitor it) ... But i can't confirm that until i receive the module (from China so ... )

rjalexa, 2016-01-08:
@Mikiya83 I have already received it (the PIR sensor) but have no clue as to how to make it work with MotionEye :-( Please come back with detailed suggestions.
Also would you be using the Fast Camera thing too to have better framerates ?
Thank you

Mikiya83, 2016-01-08:
I think i will use it like that : 
-A script in python (like this one : http://www.raspberrypi-spy.co.uk/2013/01/cheap-pir-sensors-and-the-raspberry-pi-part-1/) run in background at startup or started by another computer by SSH
-When a motion is detected (this script write a message when it occurs), another script is called with start option to start motionEye process (in init.d)
-When a motion is finished, after a delay, the same script is called with stop option

When motionEye process is started, it has to be configured to "Fast Network Camera" in advanced settings. In this mode, no motion can be set but a stream address is provided and perfs are much better, so you can read this stream on another computer and record it (when a motion is detected by PIR module, after starting motionEye process, you can send a message to this computer by SSH or HTTP to tell him to record the stream for example).

Crazyslade, 2016-01-09:
@ccrisan That worked great! Thank you! Also thanks for the explanation!
@Mikiya83  Now if I can somehow figure out how to do as you explain. Thanks for your efforts as well.

Mikiya83, 2016-01-09:
First try to monitor your PIR module with the script above.
Then when it works, instead of printing text, try to execute a little simple SH script whit it.by "subprocess.call('bash script here', shell=True)"
Then the tricky part : if you want to start/stop all motionEye process on detection, your SH has to start/stop the motionEye init.d script; if you want to monitor only your recording, it depends on your installation. For me, i use a QNAP NAS as recording system, so it always reads  the stream and i have to send him a specific HTTP request to start recording.If you are stuck somewhere, feel free to asking help :)

niallie, 2016-01-19:
I'll chip in with my settings, because I've also been struggling with this issue. I've now got it working where motion detection functions and video recording is at 10fps, with a Pi2B and a Pi Camera Module.

Services: All FTP ON | Samba Off | SSH Off
Fast Network Camera: OFF
Video Resolution: 640 x 480
Frame Rate: 10fps
File Storage: USB Key 16GB plus upload to Google Drive
Video Streaming: OFF
Still Images: OFF
Movies:
Quality - 25% | Motion triggered | Minimum length 30 seconds (doesn't work!) | Preserve for One Week
Motion Detection: 1.5% | Auto Noise Detection: OFF | Noise level 12% | Motion Gap 10s | Captured Before 0 | Captured After 100 | Minimum Frames 10
Notifications: OFF
Schedule: Inactive

I think the key thing is to keep the resolution low and try not writing stills at the same time as movies.

JahFyahh, 2016-08-26:
I am really late to the party, but what i do not get is what command should/could I use to manually (or by script) start/stop MotionEyeOS recording. @Mikiya83  you send a specific HTTP request to start recording, could you explain this a little more please?
Thank you.

bortek, 2017-02-23:
Anyone willing to comment more on this Issues or can it be closed ?

tkvu2017, 2017-03-13:
hi, can anyone help me with motioneyeos. Just install the software on ubuntu-mate os but don't see the MotionEyeOS recording at all.  attachment is my motioneye.conf file...
[motioneye.txt](https://github.com/ccrisan/motioneyeos/files/837429/motioneye.txt)

closed 2017-03-13 by bortek

jazzdrumteacher, 2017-09-02:
Hi All;
I need help using Python code with my new Raspberry pi 3. My goal is to use the PIR to trigger when motion is detected, Then to turn my Camera module V2 on to take motion, I want to continue taking an image as long as their is motion. Can I send a text to my iphone 5 and store the pictures on a external hard drive?  I purchased a 1 Tera Byte and 2 Tera Byte Western Digital Hard drive which connects directly to the USB. Therefore, I am not sure I can store images. 
Thank you all!