motioneye-project/motioneyeos #110: N00b Request - Disable Camera Module LED

mehtaman, 2016-01-06:
Hi,

I am new to the Linux scene, I've setup MotionEyeOS in my RPi2B using the Rasp. Camera Module.

All is working well (Thanks to a fantastic OS) but I would like to disable the LED so I can place the camera near my front door without people knowing it's there. 

Is there a command I can run over SSH to disable the LED?

Thanks in advance!

Mikiya83, 2016-01-06:
Hi
If it's an official Pi camera, look in the advanced settings, there is a toggle to disable LED.

mehtaman, 2016-01-06:
Thanks you Mikiya83!!!

Colbey, 2016-01-11:
I have an official Pi NoIR camera and do not have this option in advanced. I used to have in on motionPi but not since upgrading to motionEyeOs. is there something wrong with my config or is the Pi NoIR camera handled differently in this new version?

jaytay79, 2016-01-15:
@Colbey, did you restore a motionpi config file onto motioneye? I had the same situation as you when I tried this so wiped it and started afresh.

bortek, 2017-02-23:
This is what I do when I do not want to see lights coming from LEDs either on Pi itself or on Pi camera. Take a small peace of kids playing clay or a chewing gum and put it on the LED. Done.

closed 2017-02-23 by bortek