motioneye-project/motioneyeos #111: Advice on power from real life: arp broadcast problems and strange behaviors.

rjalexa, 2016-01-07:
Each RPi has slightly different power circuit design. My RPi 2 rev B has a polyfuse set after the microUSB and before the rest which will limit power to 1.1 A whatever the capacity of the wall PSU.

So it does not make any difference if you go above 1.5A when you power the RPi through the microUSB port.

In my case the original Raspberry foundation PSU yields 2A and then the polyfuse limits that to 1.1A.

Also note that the RPi-2 mod B allows 600mA or 1.2A downstream to the USB ports (1.2 with a firmware setting. Not sure as to how to activate it).

The RPi-2 B itself will draw around 550mA , the RPi camera module will draw 250mA, the Edimax or TPLink WiFi dongles I use draw around 300mA and so we're running close.

Apparently my setup worked (after removing the additional load of a USB SSD HDD) with occasional reboots due to lost connectivity.

The weirdest behavior was about the wifi connection not responding to ARP broadcast requests, so I needed to ping from the RPi to my other computer before I could reach the RPi via the browser (took a little to figure out and found some other reports around the same problem by googling).

All of this disappeared by using a powered USB hub and all works smoothly now.

So if you have strange connection problems, even if only with the RPi camera and a wifi dongle, please try a powered hub !

Reported just to spare some misery to anyone else searching for this problem :-)

rjalexa, 2016-01-07:
Reported just to spare some misery to anyone else searching for this problem :-)

closed 2016-01-07 by rjalexa

ccrisan, 2016-01-07:
Thanks for sharing!