motioneye-project/motioneyeos #112: 8192cu based Wifi dongles don't always respond to ARP broadcast requests

rjalexa, 2016-01-07:
Not really a MotionEyeOS problem but rather a buggy driver it seems ...

I have seen this with both an Edimax and a TP-Link dongles, even after powering them from a standalone powered USB hub.

From times to times the MotionEyeOS Raspberry will not respond to any connection attempt and if you check on the remote host with the arp command (tried from both a Mac OS and a Raspbmc linux box) you'll see that the MAC address for the MotionEyeOS wifi interface is not known.

The only thing you can do is to ping from the MotionEyeOS towards the machine you want to use to browse and then the MAC address will be populated and therefore all will work as advertised.

Where could I put a little cron job or similar thing which will send a ping packet to a desired machine to do this automatically ?

ccrisan, 2016-01-07:
https://github.com/ccrisan/motioneyeos/wiki/Tweaks#running-commands-at-regular-intervals-cron-support

rjalexa, 2016-01-07:
Thank you very much.

closed 2016-01-07 by rjalexa