motioneye-project/motioneyeos #113: Add 1080p resolution choice - (1920 × 1080)

chaseacton, 2016-01-07:
I have a camera that supports 1080p, but I do not see that as a choice.

Crazyslade, 2016-01-11:
I did not see that option at first either.  Try the Fast Network Camera option in the settings.  Then you should see the 1080p option in the drop down list.

Chiny91, 2016-01-11:
Alternatively, you could edit the configuration file directly as per post https://github.com/ccrisan/motioneyeos/issues/62#issuecomment-166105988 .

chaseacton, 2016-01-11:
Thanks guys, but neither of the above are great solutions. Fast Network Camera disables all motion detection and recording, so what's the point? Editing the config file works, but will get overwritten in future updates. This needs to come in the form of an updated config file in a future release.

Chiny91, 2016-01-12:
In practice, config files are **not** overwritten for tweaking *width * height*.

If you require 1920 * 1080 for all of your cameras (or just a single camera), then you can use the motionEyeOS GUI to add width * height as *Extra Motion Options* - no overwriting.

It is only if you require that 2 or more cameras have different width * height, these need be added to the thread-**N**.conf files directly and these extra .conf lines do not get overwritten with motionEyeOS updates nor by mods done via the GUI, all as per the *Tweaking motion* section of the wiki.   Certainly I have not needed to keep re-writing the .conf files, although I do have a backup just in case.

Obviously it would be simpler for the GUI to allow per-camera resolution choices and perhaps that will get added but for now, I have found the above effective.

dmarkscouk, 2016-05-11:
I've tried adding width 1920 and height 1080 to the extra motion options input field, but the video tends to be quite jumpy when I open the recorded video file (I'm don't normally have any issues playing HD content). I've assigned 800 to the GPU memory field. :/

sprinter11, 2016-05-18:
First you need to download the source of motioneye, not motioneyeos.

For v4l2 cams in motioneye-master/motioneye/v4l2ctl.py replace the line
 `if width % 16 or height % 16: # ignore non-modulo 16 resolutions`
with
`if width % 8 or height % 8: # ignore non-modulo 8 resolutions`
 For net cams add
`(1920, 1080)` to the array COMMON_RESOLUTIONS in utils.py. Previous line should be ended by a comma.

Then compile config.py by
`python2.7 config.py` which creates utils.pyc and v4l2ctl.pyc
Finally you need to place those *.pyc in /usr/lib/python2.7/site-packages/motioneye with replacing older files and reboot the device.
And that is the result:
[image]

dmarkscouk, 2016-05-19:
@sprinter11 Thanks. Will this make the video smooth / non jumpy?

sprinter11, 2016-05-19:
@dmarkscouk this trick doesn't affect the video quality. This will only make the config file will not be overwritten with a minimum 320x240, if you manually set `height 1080` and `width 1920`

ghostyroasty, 2016-07-04:
Adding 1080 should be an easy issue to fix, but I could be wrong. I'd like the option as well for my camera. I'm using 720p for now, but the camera can do 1080

Voodoofly, 2016-08-24:
@sprinter 11  
sorry i read manny hours in WWW but im to stupid to compile the "config.py"

i downloaded the "Python  2.7.12 Shell" , is this the right tool ?

please can you help with step by step to compile it

"Then compile config.py by
python2.7 config.py which creates utils.pyc and v4l2ctl.pyc
Finally you need to place those *.pyc in /usr/lib/python2.7/site-packages/motioneye with replacing older files and reboot the device."

Thx 
Tomislav

ccrisan, 2016-08-24:
Don't bother compiling it. Just place the new config.py next to the old config.pyc and it will be used automatically.

ccrisan, 2016-08-24:
See this: https://github.com/ccrisan/motioneye/blob/master/motioneye/config.py.

Voodoofly, 2016-08-24:
i have installed motioneye 20160801 on the SD Card, is there any easy way to bring it to the resolution 1920x1080 "with Motion detection"  include ?

i put out the SD Card and read it in win10 Explorer, can not find the above files

Sorry, i think your answer didnt help me, because i am a Stainless Steel "Welder" not a programmer  :-) 

Ccrisan why you didnt implement this option, whats the problem

sprinter11, 2016-08-25:
@Voodoofly, you do not need to modify config.py. Only v4l2ctl.py and utils.py. Also, it is not necessary to modify the * .pyc files directly, because these files are not human readable.
config.py is needed only because it has v4l2ctl.py and utils.py and in dependencies. When compiling config.py its dependencies also will be compiled.
To get these files you need to download the source https://github.com/ccrisan/motioneye, don't install.
Extract the ZIP with the source and go to motioneye-master/motioneye dir. Then enter the command `python2.7 config.py` in terminal here. If no errors *.pyc files will be created.
I use Ubuntu and installed Python2.7 using software-center. You can also find it on MotionEye over SSH.

Voodoofly, 2016-08-29:
have problems to copy the files to the folder
i use the motioneye image only, not some OS like Raspian or so

i try log in as root and pi, but always the same problem
 

root files system is mounted read only :  ? 
grep "\sro[\s,]" /proc/mounts
/dev/root / ext4 ro,relatime,block_validity,delalloc,norecovery,barrier,user_xattr,acl 0 0
/dev/mmcblk0p1 /boot vfat ro,relatime,fmask=0022,dmask=0022,codepage=437,iocharset=ascii,shortname=mixed,errors=remount-ro 0 0

Voodoofly, 2016-08-29:
Just found the commands to remount the Partition (https://github.com/ccrisan/motioneyeos/wiki/Tweaks), transfered the files and re-compiled them. It's working now. Thanks.

ccrisan, 2016-11-16:
Fixed with https://github.com/ccrisan/motioneye/commit/f1298c012312aee69e231d2831abab2c3d76a0a6 and https://github.com/ccrisan/motioneye/commit/d61858b0ad47a13428e850ee475619ac57e0af62.

closed 2016-11-16 by ccrisan

vdb-sander, 2017-02-14:
It's not clear to me how to change the resolution. I'd like to have a 508x680 resolution for a school project.