motioneye-project/motioneyeos #114: Google Drive - Add option to delete after X days

chaseacton, 2016-01-07:
Similar to how you can choose how long to keep images on disk before they are deleted, it would be great to extend this feature to Google Drive.

Banditen01, 2016-01-24:
Maybe not the solution You ask for but I made script file placed in Windows auto-start folder that cleanup my local Google Drive folder (must have a dedikated folder for motionEye files) this folder auto-synchronize with Google Drive Cloud and whola files are gone. A way of going round this matter.

donnlee, 2016-12-02:
I had this need too and did the following workaround.  I use Dropbox.
I have a linux server running Dropbox. This server exports a NFS share: ~/Dropbox/motioneye/cam1/
My motioneyeOS RPi mounts this NFS share at boot time to /data/nfs/
Nice thing is: motioneye has no access to my other Dropbox subdirectories (folders), so it cannot delete my other Dropbox files (like if there was a bug).
I configure motioneye to write stills and movies to /data/nfs/
and set the retention time for stills to 2 days,
(retention time for movies: 1 day).

bortek, 2017-03-01:
One can build a lot of nitty-gritty solutions around google drive which re outside of the scope of motioneye project. For example https://github.com/odeke-em/drive that was mentioned above can easily be wrapped around  a simple script and cronjobed to do what you ask for. As of now this is something the users will have to do themself. 

Here are some discussion which can help you out and be the source of inspiration :) 

http://stackoverflow.com/questions/14628895/automatically-delete-files-in-google-drive
http://stackoverflow.com/questions/28618964/apps-script-automatically-delete-files-from-google-drive-older-than-3-days-g
http://community.ispyconnect.com/ispybb2/viewtopic.php?t=4698

I am closing this issue.

closed 2017-03-01 by bortek