motioneye-project/motioneyeos #115: Email notification yahoo--> gmail

dbemme, 2016-01-08:
Dear Crisan,
I've tested for few months previous project (motionpie) and I would like to thank you about nice project.
This week I'm testing new project (motioneyeos) and I figure out issue regarding email notification despite of same configuration. 

Server : smtp.mail.yahoo.com
Port: 587
TLS : yes
Email destination : GMAIL

In attachement  motioneye log . Did you need 

Regards.
Daniele.

[motioneye (1).txt](https://github.com/ccrisan/motioneyeos/files/82897/motioneye.1.txt)

birukun, 2016-01-09:
Are you able to log in to the Yahoo account elsewhere? Looks like you have to verify the Yahoo account first.
https://help.yahoo.com/kb/mail-for-desktop/send-email-pop-client-error-code-sln3532.html

dbemme, 2016-01-09:
Email account is already verified because with motionpie (same configuration)  works fine .

ccrisan, 2016-01-09:
What have you set for the "Email Addresses" field? Is it your exact yahoo email address?

dbemme, 2016-01-09:
Hi Crisan,
inside field "Email addresses" I've set destination gmail address, while inside SMTP account I've set yahoo account, right?

ccrisan, 2016-01-09:
That's the problem. Yahoo seems to not like the gmail address in the "from" (which is the same as the "to" field in motionEye). You could try to add your gmail address to your yahoo account somehow (as a secondary email address or so). Otherwise, try to use googles SMTP server.

dbemme, 2016-01-10:
Did you change something during porting from motionpie to motioneyeos system ?

regards.
Daniele.

ccrisan, 2016-01-10:
Yes, the *from* field is now the same as the *to* field. Many (other) SMTP servers complain otherwise.

LanceSaintLuke, 2016-02-19:
Is there a readme that covers the email setup?

I tried using my gmail and then got an email from google:
"Someone just tried to sign in to your Google Account from an app that doesn't meet modern security standards."

ccrisan, 2016-02-21:
The "From" field is now configurable in the UI. The change will be available starting with 0.31.

closed 2016-02-21 by ccrisan