motioneye-project/motioneyeos #116: SMS Notification

laurentli, 2016-01-09:
Hi Calin Crisan. First, thank you for your work. I discover Raspberry and your distribution, it's so amazing and powerful. I have a question, I test your email notification with motionpie, it work great. Now i want to have SMS notification. Do you have that function? with for example this module:

http://www.ebay.fr/itm/Sim800-Expansion-Board-With-GSM-GPRS-SMS-Function-For-Raspberry-Pi-2B-B-/141855363007?hash=item21073d0fbf:g:TlkAAOSwUdlWcwAk

chaseacton, 2016-01-12:
You could do this without any additional hardware. Just use the webhook feature to post to a page on your server and then use Twilio to send yourself a text message. Probably wouldn't take more than 20 lines of code.

laurentli, 2016-01-17:
I would hardware, because i don't want to manage server and QOS... let this task to IAP (FAI). Just buy Sim card. Is it possible with Crisan project >>> motioneyeos??? Thx

ccrisan, 2016-01-17:
@laurentli, no, SMS is not possible with motionEyeOS and I don't have plans to implement such a feature.

laurentli, 2016-01-17:
thx for reply, the only way is webhook thx

jamie3, 2016-06-18:
You can use twilio API for sending SMS

closed 2017-02-23 by bortek