motioneye-project/motioneyeos #117: EAP TLS support for wpa_supplicant

dadam428, 2016-01-10:
Hi Calin!

Is there any plan for eap tls support for wpa_supplicant on your roadmap?

brgds,
Adam

ccrisan, 2016-01-10:
Not really. I mean if you know your (additional) wpa_supplicant parameters, you could add them manually to /data/etc/wpa_supplicant.conf and it should leave them untouched. If that's not the case, I'd be happy to fix it.

dadam428, 2016-01-10:
Hi,

Thanks for the answer.
I already tried that, I have changed the wpa_supplicant.conf

 >cat /etc/wpa_supplicant.conf
ctrl_interface=DIR=/var/run/wpa_supplicant
update_config=1

network={
     ssid="ssid"
     scan_ssid=1
     key_mgmt=WPA-EAP
     pairwise=CCMP
     group=CCMP
     eap=tls
     identity="id"
     ca_cert="/etc/ssl/private/ca.crt"
     client_cert="/etc/ssl/private/cert.crt"
     private_key="/etc/ssl/private/private.key"
}

and I tried to start the wpa_supplicant manually, with wpa_supplicant 
-Dnl80211 -iwlan0 -c/etc/wpa_supplicant.conf command
but it says:

Line 10: unknown EAP method 'tls'
*You may need to add support for this EAP method during wpa_supplicant**
**build time configuration.*
See README for more information.
Line 10: failed to parse eap 'tls'.
Line 15: failed to parse network block.
Failed to read or parse configuration '/etc/wpa_supplicant.conf'.

so that's my problem :)

thanks,
Adam

ccrisan, 2016-01-10:
Alright. I'm going to add support for EAP in wpa supplicant with the next build. Let's hope that's enough for your wifi connection to work.

dadam428, 2016-01-10:
Sounds good, thanks!
Is there any ETA for the new build?

ccrisan, 2016-02-21:
I added EAP support for wpa_supplicant. I hope to get the next prerelease ready within a week.

closed 2016-02-21 by ccrisan