motioneye-project/motioneyeos #121: USB hdd not selectable in File storage dropdown menu (anymore)?

olliraa, 2016-01-12:
Running the latest version with rpi2. Earlier when I tested the software for the rist time usb drives worked perfectly. Now no drive shows upin the menu (tested with many different ones). The menu only shows the "SD card" and "Custom path". lsusb via ssh lists the devices ok, but for some reasion the menu in the gui does not get updated. What is the path to usb drives/workaround for this?

ccrisan, 2016-01-12:
Please attach your log files.

olliraa, 2016-01-12:
Here are (all?) the logs attached as zip.
[logs.zip](https://github.com/ccrisan/motioneyeos/files/87553/logs.zip)

ccrisan, 2016-01-12:
Is there any reason why you left out dmesg.log (actually the most important one in this case)?

olliraa, 2016-01-12:
Sorry, took a while to figure out that I had to:

mount the root as writable
dmesg > dmesg.log

But here it is. Hope it's ok :)

-Olli

ccrisan, 2016-01-12:
Wow I see now why you ommitted it :) You could easily download it from the UI (Expert Settings).

ccrisan, 2016-01-12:
Nevertheless you forgot to attach it.

olliraa, 2016-01-12:
It was attached, but apparently some mail server virus scanner did not like
it... Let's try again :)

https://onedrive.live.com/redir?resid=C769995EB0B2279B!45495&authkey=!ABoZu-TM_lKH6uw&ithint=file%2clog

Works now? :)

-Olli

ccrisan, 2016-01-12:
Nope, still not. Are you able to see the attachment in your comment?

ccrisan, 2016-01-12:
Ah, nevermind. Just never use the mail to attach files. Always use github's UI.

olliraa, 2016-01-12:
Ok, here's the dmesg.log as text (just to make sure):

[    0.000000] Booting Linux on physical CPU 0xf00
[    0.000000] Linux version 4.1.13-meye (ccrisan@supermicro) (gcc version 4.8.5 (Buildroot 2015.08-rc1-g3394b0c) ) #1 SMP PREEMPT Sat Jan 9 20:30:46 EET 2016
[    0.000000] CPU: ARMv7 Processor [410fc075] revision 5 (ARMv7), cr=10c5387d
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] Machine: BCM2709
[    0.000000] cma: Reserved 8 MiB at 0x36800000
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] On node 0 totalpages: 225280
[    0.000000] free_area_init_node: node 0, pgdat 8058a340, node_mem_map b6112000
[    0.000000]   Normal zone: 1760 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 225280 pages, LIFO batch:31
[    0.000000] [bcm2709_smp_init_cpus] enter (93c0->f3003010)
[    0.000000] [bcm2709_smp_init_cpus] ncores=4
[    0.000000] PERCPU: Embedded 10 pages/cpu @b60e5000 s11136 r8192 d21632 u40960
[    0.000000] pcpu-alloc: s11136 r8192 d21632 u40960 alloc=10*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 223520
[    0.000000] Kernel command line: dma.dmachans=0x7f35 bcm2708_fb.fbwidth=656 bcm2708_fb.fbheight=416 bcm2709.boardrev=0xa01041 bcm2709.serial=0x79eb822c smsc95xx.macaddr=B8:27:EB:EB:82:2C bcm2708_fb.fbswap=1 bcm2709.disk_led_gpio=47 bcm2709.disk_led_active_low=0 sdhci-bcm2708.emmc_clock_freq=250000000 vc_mem.mem_base=0x3dc00000 vc_mem.mem_size=0x3f000000  dwc_otg.fiq_fix_enable=1 console=tty1 root=/dev/mmcblk0p2 rootfstype=ext4 rootwait ro rootflags=noload panic=10 quiet loglevel=1
[    0.000000] PID hash table entries: 4096 (order: 2, 16384 bytes)
[    0.000000] Dentry cache hash table entries: 131072 (order: 7, 524288 bytes)
[    0.000000] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes)
[    0.000000] Memory: 878848K/901120K available (4003K kernel code, 235K rwdata, 1228K rodata, 172K init, 359K bss, 14080K reserved, 8192K cma-reserved)
[    0.000000] Virtual kernel memory layout:
                   vector  : 0xffff0000 - 0xffff1000   (   4 kB)
                   fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
                   vmalloc : 0xb7800000 - 0xff000000   (1144 MB)
                   lowmem  : 0x80000000 - 0xb7000000   ( 880 MB)
                   modules : 0x7f000000 - 0x80000000   (  16 MB)
                     .text : 0x80008000 - 0x80524034   (5233 kB)
                     .init : 0x80525000 - 0x80550000   ( 172 kB)
                     .data : 0x80550000 - 0x8058afc8   ( 236 kB)
                      .bss : 0x8058d000 - 0x805e6ee4   ( 360 kB)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] Preemptible hierarchical RCU implementation.
[    0.000000] NR_IRQS:608
[    0.000000] Architected cp15 timer(s) running at 19.20MHz (phys).
[    0.000000] clocksource arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x46d987e47, max_idle_ns: 440795202767 ns
[    0.000014] sched_clock: 56 bits at 19MHz, resolution 52ns, wraps every 4398046511078ns
[    0.000033] Switching to timer-based delay loop, resolution 52ns
[    0.000215] Console: colour dummy device 80x30
[    0.000409] console [tty1] enabled
[    0.000434] Calibrating delay loop (skipped), value calculated using timer frequency.. 38.40 BogoMIPS (lpj=19200)
[    0.000457] pid_max: default: 32768 minimum: 301
[    0.000596] Mount-cache hash table entries: 2048 (order: 1, 8192 bytes)
[    0.000615] Mountpoint-cache hash table entries: 2048 (order: 1, 8192 bytes)
[    0.001687] CPU: Testing write buffer coherency: ok
[    0.002168] CPU0: thread -1, cpu 0, socket 15, mpidr 80000f00
[    0.002188] [bcm2709_smp_prepare_cpus] enter
[    0.002273] Setting up static identity map for 0x8240 - 0x8274
[    0.008478] [bcm2709_boot_secondary] cpu:1 started (0) 16
[    0.008836] [bcm2709_secondary_init] enter cpu:1
[    0.008885] CPU1: thread -1, cpu 1, socket 15, mpidr 80000f01
[    0.010457] [bcm2709_boot_secondary] cpu:2 started (0) 17
[    0.010728] [bcm2709_secondary_init] enter cpu:2
[    0.010760] CPU2: thread -1, cpu 2, socket 15, mpidr 80000f02
[    0.012510] [bcm2709_boot_secondary] cpu:3 started (0) 16
[    0.012764] [bcm2709_secondary_init] enter cpu:3
[    0.012794] CPU3: thread -1, cpu 3, socket 15, mpidr 80000f03
[    0.012870] Brought up 4 CPUs
[    0.012894] SMP: Total of 4 processors activated (153.60 BogoMIPS).
[    0.012904] CPU: All CPU(s) started in HYP mode.
[    0.012912] CPU: Virtualization extensions available.
[    0.013660] devtmpfs: initialized
[    0.031949] VFP support v0.3: implementor 41 architecture 2 part 30 variant 7 rev 5
[    0.032217] clocksource jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 1911260446275000 ns
[    0.033410] NET: Registered protocol family 16
[    0.039260] DMA: preallocated 4096 KiB pool for atomic coherent allocations
[... 110 lines cut ...]
[    1.778963] Indeed it is in host mode hprt0 = 00021501
[    1.940912] usb 1-1: new high-speed USB device number 2 using dwc_otg
[    1.941057] Indeed it is in host mode hprt0 = 00001101
[    2.109644] EXT4-fs (mmcblk0p3): mounted filesystem with ordered data mode. Opts: (null)
[    2.114183] usb 1-1: New USB device found, idVendor=0424, idProduct=9514
[    2.114203] usb 1-1: New USB device strings: Mfr=0, Product=0, SerialNumber=0
[... cut ...]
[    2.642137] bcm2708_spi bcm2708_spi.0: SPI Controller at 0x3f204000 (irq 80)
[    2.643232] bcm2708_i2c_init_pinmode(1,2)
[    2.643265] bcm2708_i2c_init_pinmode(1,3)
[    2.643575] bcm2708_i2c bcm2708_i2c.1: BSC1 Controller at 0x3f804000 (irq 79) (baudrate 100000)
[    4.217425] smsc95xx 1-1.1:1.0 eth0: hardware isn't capable of remote wakeup
[    5.732505] smsc95xx 1-1.1:1.0 eth0: link up, 100Mbps, full-duplex, lpa 0xCDE1
[   11.271909] random: nonblocking pool is initialized
[ 1964.415408] EXT4-fs (mmcblk0p2): re-mounted. Opts: norecovery

olliraa, 2016-01-18:
Hi.

I hope the dmesg.log was "ok" :) If there's something I can do for testing
this usb-issue, please let me know. My system is not yet in "production
use", so I'd glad to help.

-Olli

ccrisan, 2016-01-18:
Your dmesg.log file is as good as it can be. It's my spare time that's not too abundant these days. I haven't gotten to test this yet.

olliraa, 2016-02-07:
It was all due to too low voltage for the usb. Just add max_usb_current=1 to config.txt and the usb drives work ok :)

closed 2016-02-07 by olliraa