motioneye-project/motioneyeos #122: Updating on Pi A+ failing

jaytay79, 2016-01-15:
I've had this happen to me on a few occasions now but always chalked it up to something wrong with my config but having just tried some tests I can reproduce this every time I attempt to update. I have this issue on two Pi A+ models and both use WiFi.

I click on "Check" for updates on the web UI and am informed a new update is available, the Pi starts the update process but the UI just shows the spinning "Updating" message. Looking at motioneye.log as I do the update this is what shows up:

```
2016-01-15 14:17:34: [motioneye]     INFO: updating to version 20160109...
2016-01-15 14:17:34: [motioneye]     INFO: killing motioneye init script...
2016-01-15 14:17:34: [motioneye]     INFO: stopping netwatch init script...
2016-01-15 14:17:35: [motioneye]     INFO: downloading https://github.com/ccrisan/motioneyeos/releases/download/20160109/motioneyeos-raspberrypi-20160109.img.gz...
2016-01-15 14:18:12: [motioneye]     INFO: decompressing /data/.firmware_update/firmware.gz...
2016-01-15 14:19:10: [motioneye]     INFO: reading partiton table...
2016-01-15 14:19:10: [motioneye]     INFO: extracting boot.img...
2016-01-15 14:19:14: [motioneye]     INFO: extracting root.img...
2016-01-15 14:19:58: [motioneye]     INFO: backing up /boot/config.txt
2016-01-15 14:19:58: [motioneye]     INFO: unmounting boot partition...
2016-01-15 14:19:58: [motioneye]     INFO: installing boot image...
2016-01-15 14:20:05: [motioneye]     INFO: mounting boot partition read-write...
2016-01-15 14:20:05: [motioneye]     INFO: restoring up /boot/config.txt
2016-01-15 14:20:06: [motioneye]     INFO: preparing to boot in fwupdate mode...
2016-01-15 14:20:06: [motioneye]     INFO: rebooting...
2016-01-15 14:20:08: [motioneye]    ERROR: failed to get current picture for remote camera 1 on http://192.168.1.211/config/1: request timed out
2016-01-15 14:20:08: [motioneye]    ERROR: float argument required, not NoneType
Traceback (most recent call last):
  File "home/ccrisan/motioneyeos/output/raspberrypi/target/usr/lib/python2.7/site-packages/tornado/web.py", line 1369, in _stack_context_handle_exception
  File "home/ccrisan/motioneyeos/output/raspberrypi/target/usr/lib/python2.7/site-packages/tornado/stack_context.py", line 314, in wrapped
  File "/usr/lib/python2.7/site-packages/motioneye/remote.py", line 76, in wrapper
  File "/usr/lib/python2.7/site-packages/motioneye/remote.py", line 324, in on_response
  File "/usr/lib/python2.7/site-packages/motioneye/handlers.py", line 867, in on_response
TypeError: float argument required, not NoneType
2016-01-15 14:20:08: [motioneye]    ERROR: 500 GET /picture/2/current/?_=1452867454942&_username=admin&_signature=59701f157267da4641b6e4b240066891387f230e (192.168.1.87) 154375.75ms
2016-01-15 14:20:08: [motioneye]    ERROR: mjpg client timed out receiving data for camera 1 on port 8081
2016-01-15 14:20:14: [motioneye]  WARNING: mjpg client connection for camera 1 on port 8081 is closed
2016-01-15 14:20:16: [motioneye]     INFO: interrupt signal received, shutting down...
2016-01-15 14:20:16: [motioneye]     INFO: server stopped
2016-01-15 14:20:16: [motioneye]     INFO: tasks stopped
2016-01-15 14:20:19: [motioneye]     INFO: motion stopped
2016-01-15 14:20:19: [motioneye]     INFO: smb mounts stopped
2016-01-15 14:20:19: [motioneye]     INFO: bye!
Terminated
```
And then nothing, the Pi restarts but doesn't do anything, plugging in an HDMI cable shows it's stuck at the "rainbow" screen.

As I said, this happens every time I try to update on both devices.

ccrisan, 2016-01-17:
The error in the log has nothing to do with the update process. What version were you previously running?

jaytay79, 2016-01-17:
I was on 20151230 previously, however this also happened when attempting to update from 20151206 to 20151230 as well as at least one previous release (I'm afraid I can't recall which ones though). I still have the SD card that's "stuck" in fwupdate mode if that will help at all - I note some minor differences in the config.txt and cmdline.txt files from a "fresh" image:

cmdline.txt:
```
dwc_otg.fiq_fix_enable=1 console=tty1 root=/dev/mmcblk0p2 rootfstype=ext4 rootwait ro rootflags=noload
```
config.txt:
```
start_file=start.elf
fixup_file=fixup.elf
arm_freq=900
core_freq=250
sdram_freq=450
over_voltage=0
gpu_mem=96
gpu_mem_256=96
gpu_mem_512=96
disable_camera_led=1
gpu_freq=250
initramfs fwupdater.gz
```

ccrisan, 2016-01-17:
The arm_freq=900 is definitely not ok for an A+. Have you explicitly overclocked your motionEyeOS?

jaytay79, 2016-01-17:
Yes, I've previously had both set to 900.

ccrisan, 2016-01-17:
Please try upgrading while overclocking is disabled. My A+ won't boot at all when above 800 MHz.

closed 2016-02-20 by ccrisan