motioneye-project/motioneyeos #126: Recommendations for an outdoor waterproof enclosure

paulmaunders, 2016-01-18:
Apologies if this is too off topic, but given your expertise in working with the single board CCTV systems, I wondered if there are any particular enclosures you would recommend to create an outdoor waterproof camera running your software?

I am particularly interested in a Raspberry Pi based system.

Banditen01, 2016-01-19:
Why don't you just use a regular outdoor cam?

ektus, 2016-01-19:
Using a regular outdoor cam has pros and cons:
Pro:
- Weather proof
- Small form factor
- PoE available
- Night vision with integrated IR LED
Con:
- High resolutions quickly lead to very high network bandwith that's too much for the RPi
- Using the in-built motion detection of the camera is impossible with Motioneye (yet?)
- Building your own system gives you more control
- Where's the fun in just going the easy way?

I've got a RPi2B with camera module (12mm mount, Amazon B00JM94WCW), 1TB hard disk and PoE adapter (TP-Link TL-PoE10R) installed in a fake surveillance camera housing (Elro CS66D). The housing is made from aluminum and big enough for the components, but the wall mount arm is T-shaped and not stiff enough. The casing will oscillate and thus produce false motion. I wouldn't install it directly in the rain either, but outside under a roof should be okay. I haven't got long-term results yet, the system has been installed just two weeks ago.

Regards
Ektus.

paulmaunders, 2016-01-19:
I saw a project called RainBerry, which looks pretty cool, but you have to order all the parts individually and get it 3D printed, it would be nice if someone offered a kit version...

https://tinkererblog.wordpress.com/2015/07/28/how-i-designed-a-compact-weatherproof-raspberry-pi-case/

Banditen01, 2016-01-19:
This type of box in IP65 you can buy from a electrician shop.

Jbdahan, 2016-01-20:
check this out 
http://smile.amazon.com/Monoprice-106687-11-Inch-Outdoor-Housing/dp/B009GUT3ZS/ref=sr_1_1?ie=UTF8&qid=1453326690&sr=8-1&keywords=outdoor+camera+housing

bakecroissaint, 2016-10-20:
Have you seen new waterproof enclosure from www.s7.io for RaspberryPi? You could use transparent cover for the camera, make sure it's in a separate case to allow your IR correctly. NoIR should work out of the same box. Mounting screw holes there already.

ektus, 2016-10-22:
Nice enclosure, but way too small to fit all the components I've got in mine.

bakecroissaint, 2016-10-22:
How much components have you got? What would be approximate size box you are looking for?

ektus, 2016-10-22:
That information has already been given in post number 3 in this bug, bit here again:
I've got a RPi2B with camera module (12mm mount, Amazon B00JM94WCW), 1TB hard disk and PoE adapter (TP-Link TL-PoE10R) installed.
I'm currently not looking for a new box. The existing one has been in operation since January, 2016 and has been working fine after building a much sturdier mounting arm.

bortek, 2017-03-01:
Yeah this was really off topic. :) Please have a look at your new forum where you can discuss such questions with other members. 

https://groups.google.com/forum/#!forum/motioneye

closed 2017-03-01 by bortek