motioneye-project/motioneyeos #128: How to auto-mount NAS from network

Banditen01, 2016-01-22:
Have tried ang searched for an answer,quite new to this can somebody guide me? Thx in advance.

ccrisan, 2016-01-22:
https://github.com/ccrisan/motioneyeos/wiki/Troubleshooting#network-shares

Banditen01, 2016-01-22:
Have already tried this in all different combinations but don't managed to get it work. \\nas\usb\bcl\infart or \\192.168.0.170\usb\bcl\infart is the directory, user and passwd filled in but no luck only thing that happens is that camera fails. Thx anyway for great work and patience with us beginners. Made a donation today 👍

ccrisan, 2016-01-22:
Thanks for your donation! For a path like "\192.168.0.170\usb\bcl\infart", the parameters would be as follows:

 * Network Server: "192.168.0.170"
 * Share Name: "usb"
 * Share Username & Password: your credentials, if required
 * Root Directory: "/bcl/infart"

If these still won't work, would you be able to set up remote SSH access for me to your motionEyeOS unit so I can run some tests?

Banditen01, 2016-01-22:
Because i messed up the origin storage string I made a system restore from a backup file I made, now it worked to connect to NAS

Thx

closed 2016-01-22 by Banditen01