motioneye-project/motioneyeos #131: Dropbox and Google drive

Arnaud31500, 2016-01-23:
Hi,

Congratulations for your excellent OS. It works quite well !
I'm trying to make it send my pictures to my dropbox or google drive but can't figure how to do this with the new "Upload media files" section (I tried with success the method found in https://github.com/ccrisan/motioneyeos/issues/4, but it's terribly slow...). Could you tell us how to configure it to make it work, from the beginning ? What's the location of the files ?

Thanks in advance,

Arnaud

ccrisan, 2016-01-24:
What's wrong with he "official" method? Does it give you an error? Does it say anything? Normally you'd need to click the link to obtain an authorization key, enter the (existing) directory name, test the setup and click apply.

Arnaud31500, 2016-01-25:
Hi Ccrisan,

Thanks for your prompt reply. Actually I don't understand where the directory must be located and how it should be setup. Do I have to create it by myself with mkdir on a place I choose ? Can it be the "output" directory ?
Thanks for your recommandations.

ccrisan, 2016-01-25:
You must create that directory (or path of directories) in your google drive. You can call it whatever you want.

Arnaud31500, 2016-01-25:
Thanks for your help, it works perfectly now and it's really fast :). Maybe you should precise in the help spot  that the folder has to be created inside the dropbox or gdrive account ?

ccrisan, 2016-01-25:
Will do.

closed 2016-01-25 by ccrisan