motioneye-project/motioneyeos #133: Email notification doesn't work

Arnaud31500, 2016-01-24:
Hi,
I can't get email notifications to work. When I take a look at the motioneye.log file, here's what I get:

2016-01-24 09:14:59: [sendmail]    ERROR: failed to send mail: (535, '5.7.8 Username and Password not accepted. Learn more at\n5.7.8  https://support.google.com/mail/answer/14257 cv10sm13446344wjb.17 - gsmtp')
Traceback (most recent call last):
  File "/usr/lib/python2.7/site-packages/motioneye/sendmail.py", line 199, in on_message
  File "/usr/lib/python2.7/site-packages/motioneye/sendmail.py", line 57, in send_mail
  File "/usr/lib/python2.7/smtplib.py", line 622, in login
SMTPAuthenticationError: (535, '5.7.8 Username and Password not accepted. Learn more at\n5.7.8  https://support.google.com/mail/answer/14257 cv10sm13446344wjb.17 - gsmtp')
2016-01-24 09:17:30: [sendmail]    ERROR: failed to send mail: (535, '5.7.8 Username and Password not accepted. Learn more at\n5.7.8  https://support.google.com/mail/answer/14257 z137sm10540414wmc.8 - gsmtp')
Traceback (most recent call last):
  File "/usr/lib/python2.7/site-packages/motioneye/sendmail.py", line 199, in on_message
  File "/usr/lib/python2.7/site-packages/motioneye/sendmail.py", line 57, in send_mail
  File "/usr/lib/python2.7/smtplib.py", line 622, in login
SMTPAuthenticationError: (535, '5.7.8 Username and Password not accepted. Learn more at\n5.7.8  https://support.google.com/mail/answer/14257 z137sm10540414wmc.8 - gsmtp')

I'm sure everything is good with my username and password... My configuration is:
SMTP : smtp.gmail.com
username : myemail@gmail.com
password : my password
port : 587
TLS on

Any idea of what happens ?

Thanks in advance,

Arnaud

Mikiya83, 2016-01-24:
Hi
Is your account configured with dual factor authentification ? If you have enable this feature, you have to request an "application password" in your account settings (your general password cannot work without second authentification).

Arnaud31500, 2016-01-25:
Hi
Thanks for your reply. Dual factor auth is not enabled on this gmail account, that's why I dont understand this log file and thought there could be a software sided problem. I tried with other email accounts (not gmail), it didn't work too.

Mikiya83, 2016-01-25:
It's not a motionEye problem, i use GMail notifs and it work like a charm.

Arnaud31500, 2016-01-25:
I'm pretty sure of it too, but still looking for the solution :)

Mikiya83, 2016-01-25:
Have you ever tried to connect to your Gmail account with another soft (like thunderbird,...) ?

Arnaud31500, 2016-01-25:
Yes, and it works like a charm, that's why I really don't understand what happens...

ccrisan, 2016-01-25:
Do you have any special characters in your password? That's the only explanation for this issue then.

Arnaud31500, 2016-01-25:
Yes, I do, there is a "%"...

Arnaud31500, 2016-01-26:
Hi Ccrisan,
You were right : removing special characters in my password helped : everything works ! Is there anything you can do to enable special characters in passwords ?

Thanks a lot for your help and your excellent work.

bazcurtis, 2016-02-14:
If it helps anyone, my setup for Gmail was an application specific password. I used MFA on my account.

SMTP server - smtp.gmail.com
Port - 587
TLS - On

Worked fine with that setup.

ccrisan, 2016-02-20:
Fixed upstream: SMTP passwords can now contain percent characters.

closed 2016-02-20 by ccrisan

isibbot, 2017-03-01:
If this helps anyone, If your setup with 2FA and you get an incorrect password error, setting up an App password for you gmail worked for 
https://myaccount.google.com/apppasswords

dejonghe, 2018-03-10:
As above if this helps anyone, I had to allow "Less Secure Apps": https://myaccount.google.com/lesssecureapps

joaomamede, 2019-12-24:
I just enabled Less Secure Apps, apppasswords reports I can't have it in my account.
I still have this error:
"Notification email failed: 5.7.8 Username and Password not accepted. Learn more at 5.7.8 https://support.google.com/mail/?p=BadCred........"

Zombie-aaargh, 2023-11-25:
I'm getting the same problem on a non-gmail account with no special characters in the password.