motioneye-project/motioneyeos #137: Looking for mor infos

chfakht, 2016-01-26:
Hi thanks Mr @ccrisan for you great work 
i want to use my pi for video-surveillance and i have simply installed motioneyeos . Now i want to know some info about the settings :+1: 
1. what does this options do in detail : Motion Keep-alive
2. I will use this for motion detection ONLY , please what should i put in GPU Memory .
3. Since i will increase memory i have to Overclock the gpu to get desired performance ? or what is the risk from overclocking the GPU since the pi will be used 24/24 for video-surveillance
4. Does the pi camera (the one with CSI) have auto-brightness ? or should i enable this options : however what it's side effects ?
5. How video-rotation works  , that it consume more cpu if enabled  ?
6. Motion Optimization when enabled make cpu usage decrease , right ?
7. When motion is detected i get a video of 33 secondes , where i can confgure this to give it more time
8. i didn't understund well this : Minimum Motion Frames

I'm sorry for all this questions but it's the first time that i use a motion detection software :last_quarter_moon_with_face: ...
I have another question that is the more important for me : from your experience what the parameters should i use that are optimal in term of : GPU Memory, Overclocking, Video Resolution, Frame Rate ... thanks a lot :+1:^ :100:

ccrisan, 2016-01-27:
> 1. what does this options do in detail : Motion Keep-alive

Where have you seen this one?

> 2. I will use this for motion detection ONLY , please what should i put in GPU Memory .

The minimum I believe would be 96 MB. I recommend 128 MB.

> 3. Since i will increase memory i have to Overclock the gpu to get desired performance ? or what is the risk from overclocking the GPU since the pi will be used 24/24 for video-surveillance

I recommend using a RPi2, as the RPi1 might not have enough processing power for higher resolutions. Overclocking would give you a boost of ~20% at best. I don't know about the risk, honestly.

> 4. Does the pi camera (the one with CSI) support auto-brightness ? because when i have enabled this options : brightness goes up and down , not working ?

Auto-brightness in motion(Eye) is software-based and often doesn't work well. There might be a camera-specific one but I couldn't tell you more on this matter.

> 5. How video-rotation works  , that it consume more cpu if enabled  ?

I doubt the rotation overhead would be significant.

> 6. Motion Optimization when enabled make cpu usage decrease , right ?

It pauses the streaming if no motion is detected. It probably does.

> 7. When motion is detected i get a video of 33 secondes , where i can adjust this to give it more time

The duration is given by the number of seconds of detected motion. You should adjust the motion detection parameters if you think your setup misses motion frames.

> 8. i didn't understand well this : Minimum Motion Frames !!!

http://www.lavrsen.dk/foswiki/bin/view/Motion/ConfigOptionMinimumMotionFrames

chfakht, 2016-01-27:
Thanks you've helped me , please can you tell me what i have to install to use motionEyeOS (from lamp to everything) because i have just reinstalled ubuntu 14.04

ccrisan, 2016-01-28:
Ah, you're right. The keep-alive option will constantly keep talking to motion, having MJPG clients permanently connected and ready to serve JPEG frames. It controls [this](https://github.com/ccrisan/motioneye/wiki/Configuration-File#mjpg_client_idle_timeout) setting.

closed 2016-01-28 by ccrisan