motioneye-project/motioneyeos #138: Question about the recorded video

dutchronnie, 2016-01-26:
Hello,

I have a question, i don't know if this is the right place, but i have been searching for a forum but can not find one. 
when there is motion, there is a video recorded, but my video's have only 1 picture every 4 seconds.
I have changed the frame rate, but i didn't get better pictures.
How can i set this to get a smooth video?

And is it possible to add a record button?
So i can make a video or picture when i want that to do.

Thanks, Hoping to hear soon.

chfakht, 2016-01-26:
I have the same issue , have you tried to add more GPU Memory and overclocked ?

ccrisan, 2016-01-26:
If it's a RPi1 you're talking about, then you're probably using a relatively big resolution and the processing power offered by your PI is not enough (hence the frame skipping). Try reducing both resolution and frame rate and see if the problem persists. I suggest using 640x480@5fps with a RPI1.

chfakht, 2016-01-26:
@ccrisan  what about overclocking the GPU ?

ccrisan, 2016-01-26:
@chfakht it doesn't make too much of a difference. It may give you a 20% performance boost but that's not really sensible.

dutchronnie, 2016-01-26:
I am using a RPi1, and with a resolution of 640X480 and a framerate of 6, i shall change that to 5

ccrisan, 2016-01-26:
Not really. Then your problem may be different. Please attach all of your logs. Maybe I can spot something fishy. You could also post your configuration (either a screenshot or your config files) *removing sensitive data*.

dutchronnie, 2016-01-26:
I am trieing to attach the log files, but the window does not accept *.zip

dutchronnie, 2016-01-26:
[image]

ok, i have renamed the zip file to a gif, you have to rename it again to a zip.

dutchronnie, 2016-01-26:
Is there a way to manually start the record of an avi, so i can see how much frames are in it?

ccrisan, 2016-01-26:
I'm sorry but you haven't managed to attach the logs. No, there is no record button.

dutchronnie, 2016-01-26:
[image]
[image]
[image]
[image]
[image]

Now the logfiles seperate, also renamed to a *.gif

dutchronnie, 2016-01-26:
How can i send the logfiles?

ektus, 2016-01-26:
Downloading the archive https://cloud.githubusercontent.com/assets/16868762/12590319/a2d67e72-c462-11e5-95bc-5dd5d0f1f307.gif did work, I was able to save it as .zip and open same, but that's no use as I can't read much from them. I did a re-upload of the file as zip, see below.
[a2d67e72-c462-11e5-95bc-5dd5d0f1f307.zip](https://github.com/ccrisan/motioneyeos/files/105467/a2d67e72-c462-11e5-95bc-5dd5d0f1f307.zip)

dutchronnie, 2016-01-26:
Calin I Have send the log files by mail.

ccrisan, 2016-01-27:
@dutchronnie here are a few notes:

* Your boot.log shows many reboots without a proper shut down. Have you cut the power of your motionEyeOS many times? Are you using a powerful enough PSU?

* You seem to be using a network share for storage. That could pose problems if the network connection is troublesome. Please try saving movies locally (on the SD card) and see if the problem persista.

dutchronnie, 2016-01-27:
When i change some settings, Motioneyeos  wants to reboot sometimes, I have only 1 or 2 times did a reboot.
I asume that when motioneyeos wants to reboot, he does a proper shutdomn.
I am using de Rpi 1 model B
In the settings GPU = 128

Yes i am using a network share, it's a harddrive connected to the router with USB 3.0
The Pi has a wired network connection.
I will try to store the files localy.

The google drive option is not working on my system, so i can only store 32GB.

closed 2017-03-01 by bortek