motioneye-project/motioneyeos #140: How to build motionEyeOS from scratch

chfakht, 2016-01-28:
Hi , i want to build motionEyeOS from scratch (starting from buildRoot) because i want to customize it by adding some other packages that doesn't exist : let say that i want to install opencv to add some features like face detection or recognition ... can you tell me mr @ccrisan where to start ...
thanks a lot

ccrisan, 2016-01-29:
[This](https://github.com/ccrisan/motioneyeos/wiki/Building-From-Source) is a good place to start.

closed 2016-02-21 by ccrisan

jazeph, 2018-09-24:
> Hi , i want to build motionEyeOS from scratch (starting from buildRoot) because i want to customize it by adding some other packages that doesn't exist : let say that i want to install opencv to add [...]

Did you have any luck on this? I am interested in the same...