motioneye-project/motioneyeos #142: Dropbox and Google drive are not working

dutchronnie, 2016-01-29:
Hello,

How can i get the filestorage to dropbox or google drive working?

When i push the button "test service", i get the message "Accessing the upload service succeeded"
I seemed that al is alright, but my files are noy uploaded to dropbox or google drive

ccrisan, 2016-01-29:
Log files would help. Please attach them.

dutchronnie, 2016-01-29:
OK, try to upload the logs.
If it fails i send it to your gmail

[messages.TXT](https://github.com/ccrisan/motioneyeos/files/109831/messages.TXT)
[motioneye.TXT](https://github.com/ccrisan/motioneyeos/files/109830/motioneye.TXT)
[motion.TXT](https://github.com/ccrisan/motioneyeos/files/109829/motion.TXT)
[dmesg.TXT](https://github.com/ccrisan/motioneyeos/files/109832/dmesg.TXT)
[boot.TXT](https://github.com/ccrisan/motioneyeos/files/109833/boot.TXT)

Renamed them to *.txt

ccrisan, 2016-01-29:
Hi,

 I got your logs just fine. Here's what I can tell from reading them:

1. Something seems to issue some unexpected HTTP requests to your motionEye server:

    * GET /Streaming/channels/102/picture
    * GET /cgi-bin/getstream.cgi?VideoType=2
    * GET /stream.cgi?Ch2
    * GET /cmd=list/

     ... and the list could go on. Even though this is probably not the cause of your problem, you'll need to check to see what's going on.

2. The message "failed to mount smb share \\RT-AC68U/..." indicates that you have probably tried to configure your motionEye to work with a network share but you have inputted wrong parameters (there should be no blackslashes in those params).

3. Various "Connect error on fd 13: ECONNREFUSED" errors are a result of motion daemon not running correctly, which is in turn probably a result of misconfiguring the network share params.

4. "timeout waiting for the media listing process to finish" indicates that there already are too many media files and your board cannot parse them in time for sending the notification email; the email will therefore have no attached pictures.

5. "request failed: HTTP Error 409: path/malformed_path/" seems to be the cause of your Dropbox issue; What are your dropbox settings?

dutchronnie, 2016-01-29:
[image]

This are my settings for dropbox.

1. I don't know what to do about this. Is it better to do a fresh install from image?

2. i think this was one of the previous try's to connect the drive, later i get the settings correct.

3. i save the images now on the local sd card, and not to the network share.

4.You see on the disk usage that the sd card is not full of pictures

5. see picture

ccrisan, 2016-01-29:
The strange requests don't have to do anything with motionEyeOS but rather with other devices on your network that are doing those requests. I have no idea where they could come from.

Do you have the folder "MotionEyoOs" created in the root of your Dropbox? Are you sure you really meant "MotionEyoOs" and not "MotionEyeOS"?

dutchronnie, 2016-01-29:
I think i know who makes the requests, I have another pi with domoticz installed, and somewhere in the settings it searches for a camera on the ip from motioneyeos.

Yest i have the folder (MotionEyoOs) in the root of dropbox, and both names are the same, type mismatch of me :-)

dutchronnie, 2016-01-29:
I have installed a new image, and again i can not access dropbox, with a key installed.
Test service is OK.

When i choose for google drive, and ask for a key, and place the key in "autorization key" and then test the service i get the message that the folder is missing, but it is not missing, it's on google drive.
What to do next?

dutchronnie, 2016-02-01:
I have installed everything again, but still no dropbox, and no google drive.
Is there anything i can do to make this work?

ccrisan, 2016-02-21:
@dutchronnie have you managed to get it working? If not, could you please write the image from scratch, enable debugging (in Expert Settings), retry the operation and send me the logs?

dutchronnie, 2016-02-21:
No i don't have both working.

I try to write the image from scratch this week, a little bit bussy now.

dutchronnie, 2016-02-23:
I have installed the new version off motioneyeos from scratch.
But google drive and dropbox are not working.

But now i have several other problems i didn't had before
I have 2 camera's on 1 PI. the original pi camera and a usb camera.
In the previous version they worked both always. Now i don't have a picture from both camera's.
i send the log files, but i can not download the message.log

[dmesg.txt](https://github.com/ccrisan/motioneyeos/files/143331/dmesg.txt)
[motioneye.txt](https://github.com/ccrisan/motioneyeos/files/143334/motioneye.txt)
[motion.txt](https://github.com/ccrisan/motioneyeos/files/143332/motion.txt)
[boot.txt](https://github.com/ccrisan/motioneyeos/files/143333/boot.txt)

ccrisan, 2016-02-24:
Your system seems to reboot often without a proper shutdown. Check your power supply.

I'm also seeing `Failed to open video device /dev/v4l/by-id/usb-046d_081b_2549C540-video-index0: No such file or directory`. Have you written the OS image from scratch or have you updated a previous installation? Your camera device once detected is now longer available.

If you're going to use a USB camera as well as the CSI camera, my advice is to use a powered USB hub. Also make sure you haven't overclocked the system and that all the cables are properly connected.

dutchronnie, 2016-02-24:
It's a heavey power supply i use (4A), must be enough.

On both camera's i do get this messages (Failed to open video device), and don't get them back.
I have downloaded the latest image, and i have write that image to the formatted SD card, so there are no previous settings.
I have tried to resore previous settings once, but even when i didn't i get the failed video message.

I wil try a usb Hub, maybe it helps.

The only reboot i did was a push on the reboot button, or when i have changed settings and motioneye want to reboot.
I have tried to overclock with various settings, also the standard settings, but the result are the same.

I try this evening to write a new sd image from the previous version of motioneyeos.

ccrisan, 2016-02-24:
Please do and keep me posted.

dutchronnie, 2016-02-24:
I have installed a new image from the previous version of motioneyeos.
Everything went well, til i enable the debug settings.
After that everything goes wrong, no video pictures any more, and sometimes the page doesn't refresh. I get errors from the network share, before the debug settings the network share works well.

Then i have installed a new image with the latest version, don't touched the debug settings, now both camera's are working fine.
networkshare is working, but google drive and dropbox are not working.

Sinta77, 2016-03-31:
Hi dutchronnie and ccrisan, any updates on the problem of the media not being uploaded to the google drive?
Because I have the exact same problem.
Checked all the settings and logs and reinstalled from scratch already and still nothing is being uploaded.
Any advice would be helpfull.

Thanx.

joubin, 2016-04-06:
I'd like to report the same problem. I've uploaded my log files, but I should explain how the problem started.

Per the initial setup, I had configure it to upload to my google drive... I accidentally deleted the folder on google drive. Upon recreating the drive, I found out nothing was being uploaded... Here is the kicker, everything was being uploaded to the trash folder.

I got tired of trying to empty the trash. So I turn off the upload.

However, files kept showing up in the trash. So I removed motionEyes authorization from google. Created a new folder and then tried to authorize it again. Motion eye would open the google page, I would answer yes, and past in the auth code. However, motion eye kept saying the auth has been revoked. 

Giving up on google drive, I did the auth with Dropbox. It knows when the folder wasnt there when I 'tested' the service. But upon saving it, nothing shows up on dropbox. 

My 2 cents: I think something is cached somewhere. Where can I find cached location for those auths?

[boot.txt](https://github.com/ccrisan/motioneyeos/files/205640/boot.txt)
[dmesg.txt](https://github.com/ccrisan/motioneyeos/files/205642/dmesg.txt)
[messages.txt](https://github.com/ccrisan/motioneyeos/files/205641/messages.txt)
[motioneye.txt](https://github.com/ccrisan/motioneyeos/files/205639/motioneye.txt)

ccrisan, 2016-04-06:
The issue you're experiencing is thanks to the Google Drive implementation which relies on folder ids rather than on names or paths. Once your folder X has been associated with motionEye, motionEye will remember its ID and no matter where your folder is placed (e.g. Trash), files will go directly to it.

The code handling Google Drive upload should probably be rewritten to overcome these issues but I have yet yo find some spare time to fix it.

e14mattc, 2016-06-30:
+1 for getting this problem. Bit of a shame really.. everything else is working pretty good :)

closed 2017-03-01 by bortek

JamesMcMurran, 2018-06-28:
Can you link me to the code that is responsible for this? I would like to take a look at it.