motioneye-project/motioneyeos #144: Failed to build motioneyeos from source

chfakht, 2016-01-30:
When trying to compile from source , I have an error
```
./build.sh raspberrypi xconfig

mkdir -p /media/larbi/eDisque/pfa/eyeOS/motioneyeos/output/raspberrypi/build/buildroot-config/lxdialog
PKG_CONFIG_PATH="" make CC="/usr/bin/gcc" HOSTCC="/usr/bin/gcc" \
	    obj=/media/larbi/eDisque/pfa/eyeOS/motioneyeos/output/raspberrypi/build/buildroot-config -C support/kconfig -f Makefile.br qconf
  CHECK   qt
* Unable to find the QT4 tool qmake. Trying to use QT3
*
* Unable to find any QT installation. Please make sure that
* the QT4 or QT3 development package is correctly installed and
* either qmake can be found or install pkg-config or set
* the QTDIR environment variable to the correct location.
*
make[2]: *** No rule to make target `/media/larbi/eDisque/pfa/eyeOS/motioneyeos/output/raspberrypi/build/buildroot-config/.tmp_qtcheck', needed by `/media/larbi/eDisque/pfa/eyeOS/motioneyeos/output/raspberrypi/build/buildroot-config/qconf.o'.  Stop.
make[1]: *** [/media/larbi/eDisque/pfa/eyeOS/motioneyeos/output/raspberrypi/build/buildroot-config/qconf] Error 2
make: *** [_all] Error 2
```

I don't know if motionEyeOS is compatible with QT5 because i have installed QT 5.4 in /home/frano/Qt
```
qmake version
qmake --version 
QMake version 3.0
Using Qt version 5.2.1 in /usr/lib/x86_64-linux-gnu
```

Can you help please , thanks.

ccrisan, 2016-01-30:
This is a [buildroot](https://buildroot.org/) issue rather than a motionEyeOS one. I myself use QT5 and everything seems to work fine.

closed 2016-01-30 by ccrisan

chfakht, 2016-01-30:
ok i'll check the buildroot mailing list :) thanks