motioneye-project/motioneyeos #146: USB stick recording problem

alainthebest, 2016-01-31:
Hi, on RPi2 using your image software the usb drive go away after one or two days.
the usb current limitation is off.
I've configured recording on the external usb with one differnet folder for each of my two ip cameras (axis).
motion works, recording pictures and movies too but after a undefined time the recording go to the SD card an the usb stick is not viewed... reboot don't change.
I've tried differents usb sticks with the same result (16GB and 64GB)

do you have some way for finding a solution please ?

thank you

ektus, 2016-01-31:
What's in the logfiles regarding the usb stick (dmesg)? IIRC, standard setting in Raspbian for the mounting of USB media is to re-mount them read-only if there are errors, in which case the system falls back to the internal SD card. You could try an set a static mount without that option.

Regards
Ektus.

ccrisan, 2016-02-01:
Please attach your logs.

alainthebest, 2016-02-01:
Hi, here are the logs of my system. Maybe can you see the problem.

thank's
[logs.zip](https://github.com/ccrisan/motioneyeos/files/112047/logs.zip)

ccrisan, 2016-02-01:
From your `messages.log` file: 

    Feb  1 03:06:10 meye-91cf1748 daemon.notice ntfs-3g[806]: Incomplete multi-sector transfer: magic: 0x454c4946  size: 1024  usa_ofs: 48  usa_count: 1  data: 97  usn: 105: Input/output error
    Feb  1 03:06:10 meye-91cf1748 daemon.notice ntfs-3g[806]: Record 9643 has no FILE magic (0x44414142)

It's hard to tell for sure what's going on but there's clearly a problem with your USB drive. Either the filesystem is corrupt (unlikely), the drive is physically damaged or the supplied current is not enough. Try using a powered USB hub and see if the problem persists.

alainthebest, 2016-02-01:
the system is now partially crashed : no streaming but access to the web page. can't download the logs....do you have a similar system without problems ?.
I'll again erase the sdcard system and write the image. I'll try to record only on the sdcard (32GB)

scoobyd00, 2016-02-01:
Just to let you know i record to a usb stick connected to my pi2 and its been stable for over a month now. No issues at all.
Mines directly connected to my pi2, ive got a decent power supply, you could try a hub to rule out any power issues.

ccrisan, 2016-02-21:
I think this issue is fairly old and the root cause is probably hardware-related, so I'll close it now. Feel free to reopen it or post a comment if you think there's anything I can do about it.

closed 2016-02-21 by ccrisan